emoncms/emoncms #541: Issue logging input to feed: "Error could not save process list Undefined"

glynhudson, 2016-06-23:
Trying to log an input from an MQTT test input to Emoncms (local) V9.7.1 results in "Error could not save process list Undefined" when I try and save the process list setup after logging to a feed. Logging an emonPi feed (also coming from MQTT) works fine.

I think the issue is `contextid=0` and object id is 0, see console output. Any idea what could be causing this? Test input seems to be updating correctly. 

[image]

[image]

mastrogippo, 2016-08-30:
As reported on the user forum, I have the same issue. To replicate:
Post messages from the MQTT client as "emon/EVSE/power1" with an int power value.
Add this in the configuration file under [nodes]
```
[[42]]
nodename = EVSE
[[[rx]]]
names = power1
datacodes = h
scales = 1
units = W
```

The new node appears in the Inputs tab and receives the correct number
Try to add a "Log to feed" and click save:
Get this error: "ERROR: could not save processlist. undefined."
The feed appears in the Feeds page, but the value is always NaN and doesn't update.

Trickx, 2016-11-28:
Is there any progres on this issue?
Is there a known workaround?
How may I support?
I would like to publish my smartmeter data via MQTT to emoncms instead of using the rest API.

glynhudson, 2016-12-06:
Sorry for the delay trying go debug this. We have been struggling to re-create this issue reliably.

We have isolated the issue to the fact that the input is not being created in the MYQL database (which it should), but is being created in redis.  We believe the issue could be due to the MQTT input script on properly creating the input in MSQL, to try and fix this we have inserted a small delay in the the mqtt_input script https://github.com/emoncms/emoncms/commit/ba893be2fb9a2711ac6e42c844fe1e6f54b96562.

Please test is this fixes the issue for you. You will need to switch to the emoncms master branch:

```
$ rpi-rw
$ cd /var/wwe/emoncms
$ git checkout master
$ git pull 
$ sudo service mqtt_input restart
$ rpi-ro
```

JonMurphy, 2016-12-07:
so far so good!

FYI - 2nd line should be: `cd /var/www/emoncms`

glynhudson, 2016-12-07:
Thanks, been corrected.  I'll cautiously close this issue to be reopened of issue returns.

glynhudson, 2017-01-05:
Issue has not been fixed, continued discussion: 

https://community.openenergymonitor.org/t/adding-a-node-with-mqtt-problem/1396/13

glynhudson, 2017-01-15:
Interesting: I've just experienced the process list error when trying to log an input to a feed on local emoncms on my laptop (rather than emonPi). There is no mqtt or redis, the script is not running and neither mosquitto or redis-server are installed. This means that maybe the error is not being caused by the mqtt script (as thought) but rather the input process list?....

[image]

JonMurphy, 2017-01-16:
I am curious if the Input was receiving data on a regular interval e.g., every minute??  Or an Input that was inactive (e.g., Updated was n/a and Value was NaN)?

davidski, 2017-06-15:
I just experienced this issue on a freshly imaged low-write 9.8.6 | 2017.05.10 emonBase. I was injecting the default weather (current temp, 30 minute intervals, PHPFINA engine) via MQTT. Attempts to add a process to record to a feed would error out with the standard "Unknown process" message though the feed would be created. A MQTT restart followed by a redis flushall allowed me to successfully add the recording process.

Happy to assist in squashing this persistent bug!

glynhudson, 2018-06-14:
Fixed by https://github.com/emoncms/emoncms/commit/c21302ace74533cd12a605a651a0440e26b6af28

closed 2018-06-14 by glynhudson