emoncms/emoncms #544: Emoncms-docker...ready for testing!

glynhudson, 2016-06-23:
With the help of @davehun and @spikeheap from @northwalestech I have been working on an Emoncms Docker setup. Using Docker it's possible to fire up Emoncms on a bare system (assuming Docker is installed) in a couple of minutes with all the LAMP config faff taken care of.  This is great for development since you can play about with Emoncms running in a Docker container without fear of messing up your main Emoncms install. 

In the future, Docker can even be used as a deployment tool for Emoncms. In theory, it should be possible to deploy the Docker container on any server within minutes :-)

The Docker files and instructions are in the emoncms-docker repo:

https://github.com/emoncms/emoncms-docker

Please give it a try and let me know how you get on. 

The next step will be to push an Emoncms Docker container to docker-hub to allow anyone interested in Emoncms to pull down the container and be up and running with Emoncms almost immediately. 

To do: 
- Allow docker container to startup with a large amount of test data & demo dashboards etc.
- Create ability to simulated large number of inputs 
- Simulate emoncms.org multi-server
- Add MQTT
- Backup module 
- Add Redis 
- EmonHub 
- Investigate running Docker on RaspberryPi

jlm, 2016-06-29:
I'm glad you're working on this.  I ran Emoncms on Docker for a while, but it was before data containers were easy to use, so I kept losing all the data when I restarted the container.  I feel writing for Docker requires good separation of code, configuration and data.

In theory Docker provides good scaling of web applications.  When I have experimented using it for a different app, I found that getting scaling with multiple instances to work is extremely complicated and one has to understand memcached and other issues.

glynhudson, 2016-06-29:
>I feel writing for Docker requires good separation of code, configuration and data.

Agree, thus can only be a good thing! Let me know know you get on with docker. I published a blog post last week and pushed image to docker hub

closed 2016-07-26 by glynhudson