emoncms/emoncms #552: Unexpected "[" syntax error in route.php

MarkDerbyshire, 2016-08-07:
Help
I have just installed emoncms using the emoncms-stable.zip file onto my Linux web server, followed the instructions to create the MySQL database and directories, edit settings.php file etc but I am getting this syntax error when I browse to the site
"Parse error: syntax error, unexpected '[' in /home/sites/mysite.co.uk/public_html/emoncms/route.php on line 112"

The line 112 contains if (in_array($requestMethod, ['POST', 'DELETE', 'PUT'])) {

Can someone point me in the right direction as I don't know php to see what is incorrect

Thanks
--
Mark

Paul-Reed, 2016-08-07:
There is also an [unanswered post in the forum](https://community.openenergymonitor.org/t/parse-error-after-installation/778) asking the same question.

Paul

MarkDerbyshire, 2016-08-07:
Thanks.  I have just posted there and as I was writing I had an idea.  The server was running PHP V5.2.  I changed it to V5.5 and I now have the Create feed page - progress? :)

--
Mark

glynhudson, 2016-08-09:
Yes! That's progress :-)

closed 2016-08-09 by glynhudson