emoncms/emoncms #553: Default time period in Visualisation

kensru, 2016-08-16:
Hello.

I use dashboard visualisation and every time after open dashboard set needed to me time interval (in some visualisations - days, for some - weeks)

Is it possible to set in configuration for rawdata and multigraph default interval? 

Or may be make remembering last view setting for dashboard.

Thanks. You make great job!

glynhudson, 2016-08-16:
I don't think this is currently possible. Recommend using multi graph, this
will let you set a custom time period. Make sure you select floating time
when creating in your mutigraph to ensure time advances

kensru, 2016-08-16:
Thanks for your answer. In "Extra -> Visualisation -> Multigraph there are no period settings :( Am I looking for it in wrong place?

I have 3 multigraph, one is automaticly sets to daily: http://www.awesomescreenshot.com/image/1485231/296258ac0953078019d3874c1e03c8ce , 

another 2 to weakly:
http://www.awesomescreenshot.com/image/1485234/02a835dbb916a9012a2a31af514cfbce
http://www.awesomescreenshot.com/image/1485237/336190ce7c76d01cb72ea7ef5e3c696c

How could it be? I need to set it reversely :) First should show weekly, another two daily :)

pb66, 2016-08-17:
When you setup/edit the multigraph it will store what ever settings the displayed (on that setup page) graph is set to at the time, so with the settings down the left and the multigraph displayed on the right, set the time period you want using the graphs on-screen buttons before saving the settings. 

Note - If you go back to edit the interval at a later date (as you will now) the editor doesn't recognize you changing the interval alone, so trick it by just adding a letter to the multigraph name, save it, then remove it and save it again, along with the name edits it will have saved the graphs displayed settings

The reason yours ended up different is because you were probably unaware at the time that what ever the interval is when the multigraph is saved is what sticks. 

With the other visualisations you can usually change then globally by editing the code, for example I have my rawdata graphs display 1 day not a week by editing (3600000*24.0*7) to (3600000*24.0) on line 102 of Modules/vis/visualisations/rawdata.php
https://github.com/emoncms/emoncms/blob/master/Modules/vis/visualisations/rawdata.php#L102

kensru, 2016-08-17:
Pb66, Thanks!

Resaving in settings page works for me.

Unfortunately I can't change the code becouse I use https://emoncms.org :) So it will be greate to make that settings editeble in UI.

closed 2016-08-17 by glynhudson