emoncms/emoncms #559: No process description available for process ' + source feed' with id '55'

meberli, 2016-09-22:
No process description available for process ' + source feed' with id '55'.
Add a description to Module\module_name\module_name_processlist.php in process_list() function, $list[] array at the 'desc' key.

glynhudson, 2016-11-17:
Sorry, which input process are you referring to? Plese could you post a screen grab? Sorry for the slow response

JonMurphy, 2016-11-17:
Glyn - I just came across the same process issue a few days ago.  Here is a screen shot.
[image]

Also see issue #511

glynhudson, 2016-11-17:
Ah, I see it's in the virtual feeds input processing. The description for 'Source feeds' is:

> Virtual feeds should use this processor as the first one in the process list. It sources data from the selected feed.
> The sourced value is passed back for further processing by the next processor in the processing list.
> You can then add other processors to apply logic on the passed value for post-processing calculations in realtime.
> 
> Note: This virtual feed process list is executed on visualizations requests that use this virtual feed.

and the description for `+ input` is :

>Adds the current value with the last value from a feed as selected from the feed list. The result is passed back for further processing by the next processor in the processing list.

This is quite long winded. I wonder what a good description would be for `+ Source feed` ? How about simply:

> Adds another source feed.

What do you think @chaveiro ?

JonMurphy, 2016-11-17:
It should include the differences between **+ Source feed** and **+ feed**.  I stumbled across the error trying to determine why I should pick one over the other.  I could not determine the difference.

JonMurphy, 2016-11-17:
Also - all five source process have a similar error: 
    + source feed
    - source feed
    * source feed
    / source feed
    1 / source feed

[image]

TrystanLea, 2017-02-28:
Would anyone with familiarity with Virtual feeds  be able to suggest descriptions for these? Im happy to add changes into the code. Here is the line that describes "+source feed" https://github.com/emoncms/emoncms/blob/master/Modules/process/process_processlist.php#L153

TrystanLea, 2018-07-10:
This is now fixed

closed 2018-07-10 by TrystanLea