emoncms/emoncms #560: Errors on windows on administration page due to linux specific code

joyrider3774, 2016-09-27:
Hi,

Yesterday i updated my local emoncms installation to the latest version in a long time and i noticed the administration page uses linux specific code that display's errors in windows (well at least in wamp).

For example trying to read from /proc/meminfo.

Shouldn't the code detect somehow if the host it's running on is windows or linux and act accordingly ? like read only from  /proc/meminfo on linux and not on windows or if somehow possible access this data on some other way on windows.

a few examples but i think they are all related and i had more of them

```
Warning: file_get_contents(/proc/meminfo): failed to open stream: No such file or directory in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 16
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
6	0.0281	520248	system_information( )	...\admin_main_view.php:6
7	0.0984	523320	file_get_contents ( )	...\admin_main_view.php:16
```

```
Notice: Undefined index: MemTotal in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 278
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
```

```
otice: Undefined index: MemFree in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 278
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
```

```
Notice: Undefined index: MemTotal in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 279
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
```

```
 Warning: Division by zero in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 279
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
```

```
Notice: Undefined index: SwapTotal in C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php on line 283
Call Stack
#	Time	Memory	Function	Location
1	0.0020	171200	{main}( )	...\index.php:0
2	0.0258	415904	controller( )	...\index.php:156
3	0.0269	437464	admin_controller( )	...\core.php:64
4	0.0269	438056	view( )	...\admin_controller.php:26
5	0.0281	519336	include( 'C:\wamp\www\emoncmsnew\Modules\admin\admin_main_view.php' )	...\core.php:74
```

glynhudson, 2016-09-27:
Good point, however as you can probably tell most of the user base and developers run Emoncms on Linux, you may well run into other issues. This feature was added by @AndyTaylorTweet, do you have have any ideas how we could detect if Emoncms is running on windows and disable these Linux specific requests e.g. memory use?

AndyTaylorTweet, 2016-09-27:
I'll write a PR to clean that up in a few mins...

joyrider3774, 2016-09-27:
looking at the administration page there are 2 other values that might help to see if it's linux or windows.
on my windows 10 box i get this :
Server:
OS:	Windows INTELNUC
Host: 	NT (NT)

those might help to distingush between systems

Paul-Reed, 2016-09-27:
@AndyTaylorTweet - Can't see your PR, have you sent it?

AndyTaylorTweet, 2016-09-27:
Paul - Not sure what went on there - I'll do another PR for you - ETA 10 mins

Paul-Reed, 2016-09-27:
Thanks Andy, it's there now.
...and thanks for the quick support to this issue.

Paul

AndyTaylorTweet, 2016-09-27:
No problem - I was clearly being very focussed on linux on the pi when I wrote those amendments, I forget there are people who run this software on other platforms :)

Paul-Reed, 2016-09-27:
@joyrider3774 Master branch now updated with Andy's fix. Can you confirm that this resolves your issue pls.
I don't have a Windows install to test, but presents no issues for Linux installations.

Paul

joyrider3774, 2016-09-27:
@Paul-Reed no more errors on administration page with latest pull... Thanks @AndyTaylorTweet

closed 2016-09-27 by Paul-Reed

glynhudson, 2016-09-28:
Wow, that was fast! Thanks guys 👍