emoncms/emoncms #561: Can't add new feeds based on inputs (master branch)

joyrider3774, 2016-09-27:
Hi, yesterday i installed the master branch of emoncms, i'm not certain where to go for issues but one thing i noticed is that i can't add new feeds anymore that i base upon an input node. just selecting an input node and chosing "log to feed" gives me the error "SQL returned invalid insert feed id".

i traced it down in the code and what i think is happening is that when i click on the add button to add the feed `'create':function(tag, name, datatype, engine, options)` is called in `feed.js` which calls feeds.json which calls `public function create($userid,$tag,$name,$datatype,$engine,$options_in)` in `feed_model.php` and the sql statement (`$result = $this->mysqli->query("INSERT INTO feeds (userid,tag,name,datatype,public,engine) VALUES ('$userid','$tag','$name','$datatype',false,'$engine')");` ) this sql statement fails for me giving me this error:  

```
SQL-query:
INSERT INTO feeds (userid,tag,name,datatype,public,engine) VALUES ('1','Node 19','node:19:sensor.emoncms1_feedid_29','1',false,'5')
MySQL meldt: Documentatie
#1364 - Field 'processList' doesn't have a default value
```

The problem is that in `feed_schema.php` there is a column defined `'processList' => array('type' => 'text')` yet this value is never set in the above sql statement and if i look into my phpmyadmin `processlist` had no default value set.

changing the defintion to add a default value or say Null is acceptable in `feed_model.php` for processlist did not solve my problem as the database check & update kept saying the database is up to date.

i had to manually change the default value to null in my emoncms database using phpmyadmin once i specified the default value as being null i could create / save the processlist.

then sending a test value to the input where the feed is based on worked again (i used mqtt to send the value using old mqtt input listener).

as i don't know if this is related to issue  #541 or if this is a real issue / bug with the master branch, or a problem with my local installation / update i did yesterday. I thought i'd just mention this here and perhaps someone can look into my findings and decide if this a bug of some kind or some problem with updating 

i was actually looking into #541 to see if i had the same problem with the old input listener (which it seems i don't) 

i'm also not even sure what processlist in feed_model.php is supposed todo as i can't seem to see anything to add / change / create a process list for feeds.

chaveiro, 2016-10-03:
Looks like mySQL configuration is in strict mode. What distro are you using? Did you do any changes?

See https://www.euperia.com/development/mysql-fix-field-doesnt-default-value/1509

joyrider3774, 2016-10-03:
Hi @chaveiro i was using latest wamp release on windows. It uses mysql 5.7.14 and indeed it was using the default SQL mode(ONLY_FULL_GROUP_BY,STRICT_TRANS_TABLES, NO_ZERO_IN_DATE, NO_ZERO_DATE, ERROR_FOR_DIVISION_BY_ZERO, NO_AUTO_CREATE_USER, NO_ENGINE_SUBSTITUTION.) , i did not do any changes to mysql settings all default as installed by wamp on windows

It seems wamp even has options now to change this to from default to none (which is the same as putting sql-mode="" in my.ini, actually that functionality actually put's that line in the config for you using wamp if you select mode -> none).

I was unaware of this change in mysql. I can't test if it's really solved though since i edited the database to add default values for go around this issue. But to be on the safe side i switch to none (sql-mode="" mode)

I'll close this issue as it isn't related to emoncms but a configuration problem.

closed 2016-10-03 by joyrider3774