emoncms/emoncms #564: Problem with foreign currency sign saving to database

kabus1917, 2016-09-29:
When setting up a regular zoom visualisation, I tried using russian ruble sign as currency unit which didn't work. It becomes ? once dashboard saved and reloaded.
http://www.fileformat.info/info/unicode/char/20bd/index.htm
I'm using local install running on RaspberryPi. Version 9.7.2.

[image]

chaveiro, 2016-10-03:
See if your database schema is set to UTF8.

kabus1917, 2016-10-04:
I converted existing tables and default char sets on the database to utf8_unicode_ci. Now I'm able to save currency sign correctly, but it is not displayed in the graph.

> mysql> ALTER DATABASE emoncms CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.01 sec)

> mysql> ALTER TABLE app_config CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE dashboard CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE feeds CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE graph CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE input CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE multigraph CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE rememberme CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE schedule CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
> ALTER TABLE users CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;

> mysql> SHOW FULL COLUMNS FROM feeds;
> +-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
> | Field       | Type       | Collation       | Null | Key | Default | Extra          | Privileges                      | Comment |
> +-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
> | id          | int(11)    | NULL            | NO   | PRI | NULL    | auto_increment | select,insert,update,references |         |
> | name        | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
> | userid      | int(11)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
> | tag         | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
> | time        | int(10)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
> | value       | double     | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
> | datatype    | int(11)    | NULL            | NO   |     | NULL    |                | select,insert,update,references |         |
> | public      | tinyint(1) | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
> | size        | int(11)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
> | engine      | int(11)    | NULL            | NO   |     | NULL    |                | select,insert,update,references |         |
> | processList | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
> +-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
> 11 rows in set (0.07 sec)
> 

[image]

kabus1917, 2016-10-04:
I converted existing tables and default char sets on the database to utf8_unicode_ci. I'm still not able to save dashboard correctly.

mysql> ALTER DATABASE emoncms CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.01 sec)

mysql> ALTER TABLE app_config CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.04 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE dashboard CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.03 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE feeds CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 6 rows affected (0.04 sec)
Records: 6  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE graph CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.02 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE input CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.34 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE multigraph CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.35 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE rememberme CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 2 rows affected (0.29 sec)
Records: 2  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE schedule CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 0 rows affected (0.35 sec)
Records: 0  Duplicates: 0  Warnings: 0

mysql> ALTER TABLE users CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci;
Query OK, 1 row affected (0.36 sec)
Records: 1  Duplicates: 0  Warnings: 0

mysql> SHOW FULL COLUMNS FROM feeds;
+-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
| Field       | Type       | Collation       | Null | Key | Default | Extra          | Privileges                      | Comment |
+-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
| id          | int(11)    | NULL            | NO   | PRI | NULL    | auto_increment | select,insert,update,references |         |
| name        | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
| userid      | int(11)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
| tag         | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
| time        | int(10)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
| value       | double     | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
| datatype    | int(11)    | NULL            | NO   |     | NULL    |                | select,insert,update,references |         |
| public      | tinyint(1) | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
| size        | int(11)    | NULL            | YES  |     | NULL    |                | select,insert,update,references |         |
| engine      | int(11)    | NULL            | NO   |     | NULL    |                | select,insert,update,references |         |
| processList | text       | utf8_unicode_ci | YES  |     | NULL    |                | select,insert,update,references |         |
+-------------+------------+-----------------+------+-----+---------+----------------+---------------------------------+---------+
11 rows in set (0.07 sec)

[image]

chaveiro, 2016-10-04:
It is a strange error, but i've made a fix, see if its ok now.

kabus1917, 2016-10-12:
Thanks for the quick fix!
Also having issue with saving cyrillic sign to "my electric" app config.

mysql> SELECT * FROM app_config;
+--------+--------------------------------------------------------------------------+
| userid | data                                                                     |
+--------+--------------------------------------------------------------------------+
|      1 | {"myelectric":{"use":"1","use_kwh":"4","unitcost":"3.77",**"currency":""}**} |
+--------+--------------------------------------------------------------------------+

chaveiro, 2016-10-14:
Cool!
Open an issue in the my electric app project.

closed 2016-10-14 by chaveiro