emoncms/emoncms #570: 'Hints' not displayed in vis_render.js

Paul-Reed, 2016-10-08:
'Hints' are included in the code within emoncms/Modules/vis/widget/vis_render.js for vis functions, yet they fail to display in ui.

Paul

chaveiro, 2016-10-15:
Fixed with https://github.com/emoncms/emoncms/commit/7917b16a3329fcc2c287d75de0e290a3ee77ac96

Paul-Reed, 2016-10-15:
Still same here.

For example select vis 'barchart', I can't see the text "Line colour in hex. Blank is use default." which is included in optionshint.

Paul

glynhudson, 2016-11-17:
Can you post a screenshot to illustrate this issue?

Paul-Reed, 2016-11-22:
As Chaveiro comments above, this was fixed by https://github.com/emoncms/emoncms/commit/7917b16a3329fcc2c287d75de0e290a3ee77ac96 
I'm assuming the error persisted for me due to caching, but all good now.

closed 2016-11-22 by Paul-Reed