emoncms/emoncms #572: emoncms.com always says ok, even when the request is no

Frogmore42, 2016-10-10:
When I send an obviously wrong request, I still get an ok, this does not seem like a good thing.
http://emoncms.org/input/post.json2?node=3&json={power:200thisistotalgarbage

pb66, 2017-02-12:
From the ["Struggles with the EmonCMS API"](https://community.openenergymonitor.org/t/struggles-with-the-emoncms-api/3080/1) forum thread
> Silent failures: I really struggled with bulk upload and API, mostly because of some silent failures -- places where the server returned "OK", but didn't accept my upload4. The fact that this egregious data-loss bug is labelled an "enhancement request" is the main thing that convinced me to walk away from this project.

TrystanLea, 2018-07-10:
I have created an issue for this on the emoncms.org repo https://github.com/emoncms/emoncmsorg/issues/12

closed 2018-07-10 by TrystanLea