emoncms/emoncms #573: Broken delta type bar graphs in master

mattjgalloway, 2016-10-13:
There's an issue in master right now where bar graphs in the delta mode show today as a massive negative value, like so:

[image]

This I believe is due to 69ce584 (I bisected) and it's because previously the data array looked something like this:

```
[[1476140400000,438.26626586914],[1476226800000,455.49545288086],[1476313200000,461.67785644531],[1476399600000,null],[1476486000000,null]]
```

And now it looks like this:

```
[[1476140400000,438.26626586914],[1476226800000,455.49545288086],[1476313200000,461.67785644531],[1476399600000,0],[1476486000000,0]]
```

Notice the `0` instead of `null`. Null is interpreted by the graph as a gap of sorts, and is ignored. Whereas 0 is interpreted as a proper value of 0.

@chaveiro - any thoughts here? I believe you've been doing all the work around making things typed. FWIW I think that's great work, as the typing was just masking this problem. I'd rather we fix this at the graph level.

mattjgalloway, 2016-10-13:
It looks like this is actually affecting a lot of graphs, such as the apps (thanks @pb66):

[image]

And here for a PHPFINA backed feed's graph:

[image]

I suggest we revert these cast diffs. @chaveiro?

pb66, 2016-10-13:
@mattjgalloway  - The casting is an attempted fix for the numeric datatypes not being carried through to the api's returned data see #540, the casting has been altered and moved around a fair bit for a fair while now.

mattjgalloway, 2016-10-13:
@pb66 - yeh, the casting is perfect IMO. Except I think that `null` should perhaps bypass the casting. Although that does make me wonder a bit. Perhaps just the value being missing from the data array would be nicer. Both have pros and cons. Both would break existing implementations in subtle ways, unfortunately.

pb66, 2016-10-13:
[link to forum thread for completeness](https://community.openenergymonitor.org/t/latest-dev-version-display-bug-with-calculated-kwh-bar-graphs/1877?u=pb66)

chaveiro, 2016-10-14:
Help me out here, what engine is giving the problem and what API is returning the 0 data instead of the null ?

mattjgalloway, 2016-10-14:
@chaveiro - PHPFINA for sure. Not sure about others. I haven't tested myself other engines.

It's the `data.json` API - as in, the one that is used to gather the data for the bar graph visualisation shown in my first comment.

mattjgalloway, 2016-10-14:
Can I recommend we have a function that takes a time+value tuple and sanitises it? The fact that the commit causing this specific problem in this issue has casts all over smells a bit to me. Maybe a function something like this:

```
function dataPointFromTimeAndValue($time, $value) {
  $sanitisedValue = null;
  if ($value !== null) $sanitisedValue = (float)$value;
  return array('time'=>(int)$time, 'value'=>$sanitisedValue);
}
```

Still not ideal, but moves all this logic to one place.

But I don't have full context on the data model layer yet, so I might be out - ignore me if that's not a good idea!

chaveiro, 2016-10-14:
Ok should be ok, please test and let me know...

closed 2016-10-15 by chaveiro