emoncms/emoncms #589: missing feeds letters in multigraph

kraujinis, 2016-10-29:
Hi,
I install fresh cms 9.7.5 copy to host. I have problem with displaying Lithuanian letters.
In Visualizations feed name looks correctly.
[image]
When I try to look in Full screen I'm missing one letter.
[image]
The same problem in Dashboard view.

pm-cz, 2017-01-15:
Hello, possibly my pull request #629 addresses this issue

glynhudson, 2017-01-15:
Hi @pm-cz I am trying to test your PR #629 . Could you give me an example unicode to test with? I have beeing trying to use the Lithuanian character ė as used by @kraujinis above. However unlike @kraujinis I don't seem to be able to even get Emoncms to save a feed name with this character. The ė character gets replaced with a ? when the feed name is set. See example:

[image]
[image]

Therefore am I correct that the issue is not just affecting multigraph but also the main feed list? 

I have tried #629 but it does not seem to have made a difference. 

Sorry, I am quite unfamiliar with unicode characters, we don't use them often in English, however I can see the importance of Emoncms being able to handle unicode as feed name.

pm-cz, 2017-01-15:
Hello, 
the unicode sequence I have successfully used is "směr větru" (wind direction). 

But if you are getting question mark in the feed name instead, I wonder, do you have the MySQL database configured to use utf-8 characters? Because by default the MySQL database I installed emoncms in had a problem with character set being configured to latin1 and mangling all unicode characters. But emoncms.org works with these feed names (but replaces unicode in the multigraph labels).

pm-cz, 2017-01-15:
The problem is probably in the [Installation guide](https://github.com/emoncms/emoncms/blob/master/docs/LinuxInstall.md) where creating the database has following command:

`mysql> CREATE DATABASE emoncms;`

But the correct command to use UTF-8 would be

`mysql> CREATE DATABASE emoncms  DEFAULT CHARACTER SET utf8 DEFAULT COLLATE utf8_general_ci;`

or

`mysql> CREATE DATABASE emoncms  DEFAULT CHARACTER SET utf8;`

glynhudson, 2017-01-15:
Ah yes, good point. I've not tested on an emonPi which has emoncms correctly configured. I have successfully managed to use `měr větru` as a feed name and can confirm the bug reported by @kraujinis where the unicode characters are dropped when viewing fullscreen multigraph. 

I have tested you PR and it does not seem to fix this. 

[image]

## Fullscreen view 

[image]

glynhudson, 2017-01-15:
Thanks, install guide has been updated https://github.com/emoncms/emoncms/commit/34572fc568ddd4777dc6d199b1eb9d278e4d5428

Paul-Reed, 2017-01-15:
Is there a MYSQL command to update existing?

pm-cz, 2017-01-15:
@glynhudson : I am baffled by this, since it works on my local patched installation, see attached pictures.

The order in which this must be done: 
1. Apply the patch
2. Made changes to the multigraph and saved it (required since the characters are dropped in the saving routine)
3. Display the multigraph

[image]
[image]

glynhudson, 2017-01-15:
@Paul-Reed could you try and test this on your emonPi?

pm-cz, 2017-01-15:
@Paul-Reed: It should be `ALTER DATABASE emoncms CHARACTER SET utf8;`. But this will not fix the already created tables, there would have to be `ALTER TABLE xyz CONVERT TO CHARACTER SET utf8;` commands for them (or dumping them, fixing the character set and restoring them).

Paul-Reed, 2017-01-15:
@glynhudson Don't have a emonpi, but just checked the default MYSQL characterset of my emonbase and it's `latin1`.
So presumably without converting the tables, the patch won't work?

pm-cz, 2017-01-15:
Without utf-8 you would not be able to store non-ASCII letters in the database correctly (i.e. event the feed names will be wrong)

Paul-Reed, 2017-01-15:
Not having much luck with this..

- Changed database to utf8 default
- Changed all emoncms tables to utf8

Without applying the diff I see;

[image]

So after applying the diff, a reboot, and saving a new multigraph, I see a blank multigraph;

[image]

...with the browser error;

[image]

Paul-Reed, 2017-01-15:
- Just deleted my last post -
Found an issue with javascript caching causing errors.

Results to follow...

Paul-Reed, 2017-01-15:
OK.

- Update MYSQL database to utf8
- Alter all existing emoncms tables to utf8
- Without applying diff;

[image]

...after applying diff and reloading multigraph (after colour change);

[image]

I would suggest that if we merge the PR, it won't create any problems for current users with MYSQL formatted to `latin1`, but as the installation guide has been updated https://github.com/emoncms/emoncms/commit/34572fc568ddd4777dc6d199b1eb9d278e4d5428, all future installations will benefit from full utf8 support.
Existing installations who **want to** use utf8 characterset, will additionally need to convert their existing db tables to utf8.
Do you agree @glynhudson ?

glynhudson, 2017-01-16:
Yes, agree.

closed 2017-01-16 by glynhudson