emoncms/emoncms #591: mod_status leak from emoncms served as onion hidden service

beaylott, 2016-10-31:
Issue raised here:

https://mascherari.press/dark-web-diaries-maybe-dont-put-your-house-on-the-dark-web/

I dont know if this is actually an issue in current build. I am not able to reproduce trivially. Im guessing that in the above case a request to the hidden service looks like localhost access to apache. Obviously this is not normal environment. Could consider disabling mod_status.

beaylott, 2016-10-31:
Some possible solutions for users publishing emoncms as hidden service here:

https://riseup.net/en/security/network-security/tor/onionservices-best-practices#be-careful-of-localhost-bypasses

s-rah, 2016-10-31:
Hi! I wrote the article. 

Generally, this is not an issue with the application, but with the underlying server setup. As an application  think there is not much you can do apart from pointing users towards the right information (if the user, for example, hosts emoncms on an nginx server, then there is no risk of the above issue).

I would say, that the main issue worth solving in this case is moving to POST'ing API keys rather than sending them via the URL - this would minimize the risk of authentication escalations because of mod_satus leaks.

beaylott, 2016-11-01:
Thankyou for your suggestions. I am not by any measure a 'core developer' of emoncms but I will encourage them to take a closer look at this issue.

glynhudson, 2016-11-03:
Hi, thanks for your feedback. Very interesting article, it's great you are focusing attention on security issues. We are keen to improve the application. 

Am I correct in thinking that this will not be an issue if the server is running over https like https://emoncms.org?

beaylott, 2016-11-03:
I am confident this issue does not affect emoncms.org or indeed vast majority of users running emoncms in their HAN. The issue is primarily due to a localhost bypass caused by this user running it as an onion hidden service which makes mod_status public thus allowing API keys to be visible.

beaylott, 2016-11-03:
You could consider switching to POSTing the API keys, but GETing using the API keys is by-design (in my understanding) and so not sure if inconvenience caused to users is justified by this problem.

chaveiro, 2016-11-04:
POST apikey or using Authorization: Bearer type is supported since 2015.

closed 2016-11-12 by beaylott

glynhudson, 2016-11-15:
@chaveiro can be get this POST API documented in the input API helper? I can't see any reference to posting API key in the header currently documented in `/emoncms/input/api`

chaveiro, 2016-11-15:
I think somebody already did that here https://github.com/emoncms/emoncms/commit/ebd8bdcbdc5bab38e667e81c36381ab6683d75e6

glynhudson, 2016-11-15:
> I think somebody already did that here ebd8bdc

Yes, I saw that. However I did not see a mention that the API key could be
put in the headers.

chaveiro, 2016-11-15:
It's there for a long time https://github.com/emoncms/emoncms/blob/175a9257651b1df27bc1f623435435a997e252c1/Modules/input/Views/input_api.php#L8

glynhudson, 2016-11-15:
> It's there for a long time https://github.com/emoncms/emoncms/blob/
> 175a9257651b1df27bc1f623435435a997e252c1/Modules/input/
> Views/input_api.php#L8

Ah yes, wops. How did I not spot that! Thanks.