emoncms/emoncms #595: Language selecion is not working

RDCH106, 2016-11-04:
The language used is the browser default language.

Impossible to select another one.

glynhudson, 2016-11-04:
Are you referring to emoncms.org or local emoncms? Language should be working on our emonPi pre-build SD card image. If it's a local install then language will need to be installed: https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md#18-language-support

RDCH106, 2016-11-04:
No, I am refering to the main project. The PHP project that you install in a server.

**Look my example**

I have selected English as language to use in EmonCMS:

[image]

But nevertheless, I only read in Spanish because my browser is in Spanish:

[image]

[image]

I have tested the problem in Chrome and Firefox  running a server with PHP Version 5.6.19.

A few months ago [I fixed the problem](https://github.com/emoncms/emoncms/pull/484) but now It doesn't work.

I will try to fix it.

glynhudson, 2016-11-15:
Is this still an issue? 

Language selection is working ok on the emonSD pre-built SD card for RaspberryPi. Have you installed gettext: 

```
sudo apt-get install gettext
sudo dpkg-reconfigure locales
```
Then install your language and restart apache.

More info: https://github.com/emoncms/emoncms/blob/master/docs/gettext.md

e.g Spanish 

[image]

glynhudson, 2016-11-15:
Ah sorry, have I misunderstood. Is the issue that you *want* emoncms to display in English but since your browser is set to Spanish Emoncms auto displays Spanish?

RDCH106, 2016-11-16:
@glynhudson Yes, that's the problem
I'll check it out when I have time.

chaveiro, 2016-11-16:
That is strange because i can set any language in my profile and it persists it when i login independent of the browser language.

BayanN, 2016-11-17:
I am also facing the same problem. The language does not change after I select it from the 'my Account' page. I have also tried changing the language of the browser (chrome) but the language of the web site still doesn't  change.

glynhudson, 2016-11-17:
@BayanN this is a separate issue to what is being discussed here. See my comment above regarding installing gettext and locales to enable language support. The issue being discussed here is that Emoncms is apparently incorrectly picking up the browser language.

vazeri, 2017-04-20:
A way to go around this is to do what glyn said run :

> sudo apt-get install gettext
> sudo dpkg-reconfigure locales

This however changes the entire OS language

I have yet to review it completely but I believe the problem is actually in the locale.php config file line 88 Even after picking your UTF8 language, the locale.php simply establishes English as the default language in line 88 for emonCMS a quick way to go around this is just to edit this line from en_US to es_ES (after configuring locales) and you will now see everything in spanish

closed 2017-10-05 by glynhudson