emoncms/emoncms #611: Documentation Upgrading missing modularisation

bsiege, 2016-11-22:
Hi, if one uses the recommended git pull and rename directory procedure, he will find that some modules like "Dashboard" must be pulled separately. A Mention of the modularisation in Upgrading.md would be nice.

Paul-Reed, 2016-11-22:
The upgrading.md guide was really written for previous versions of emoncms, and has not been kept up to date, although some of the processes are still relevant, there is a lot missing, including the modulisation changes which occurred in v9.x.

Please feel free to have a look at the guide and suggest changes by way of a pull request.

Paul

Paul-Reed, 2016-12-10:
Closed pending documentation update contribution.

closed 2016-12-10 by Paul-Reed