emoncms/emoncms #619: Exposing fixed interval feed meta

pb66, 2016-12-04:
See forum thread [Log to Feed interval](https://community.openenergymonitor.org/t/log-to-feed-interval/1971)

The fixed interval meta such as start timestamp and interval size should be made accessible, if not displayed on the feed page directly then perhaps via a "info pop-up" from the feeds page.

At least dding it to the "get feed fields" feed api would improve accessibility by removing the need to ssh in to run commandline queries, which even that is not possible for emoncms.org users.

hagai-shatz, 2016-12-06:
You can use the undocumented API: https://emoncms.org/feed/getmeta.json?id=1
Result for PHPFINA feed is: {"interval":10,"start_time":1461310000,"npoints":65748}

TrystanLea, 2017-02-28:
I've documented this on emoncms core here: https://github.com/emoncms/emoncms/commit/f794f016a143e556d20d4c78d493ee4123c4f211 

and emoncms.org:
https://emoncms.org/site/api#feed

thanks for pointing out that it was missing

pb66, 2017-03-01:
Thanks @hagai-shatz  and @TrystanLea.

Just tried the api and found the result is different on emoncms.org to self-hosted emoncms, only emoncms returns the number of datapoints

example response from emoncms.org
> {"interval":5,"start_time":1380911750,"npoints":21499616}

example response from self hosted emoncms
> {"interval":5,"start_time":1467633890}

is that intentional? would it take much to add number of datapoints to the self-hosted version for completeness and consistency? 

[re-opened issue for concideration,of the above rather than open another issue]

TrystanLea, 2017-03-02:
@pb66 the main differences are due to emoncms.org not having buffering implemented resulting in a different npoints function. I've managed to add it in though in a slightly different way providing consistency at the API level. https://github.com/emoncms/emoncms/commit/2d7be37dd2d1e855e4fcfd965c7e2aeed7e93d87

pb66, 2017-05-23:
Finally got to test this and it works as intended, thanks @TrystanLea

closed 2017-05-23 by pb66