emoncms/emoncms #621: consider adopting predis library to remove redis php extension requirement and to remove limitation on shared linux installation

yphyph01, 2016-12-27:
see https://github.com/nrk/predis

This is a pure PHP library. As you can see the library is seeing quite a lot of use https://packagist.org/packages/predis/predis.
For performance issue comparing the the natice extension, see: https://github.com/nrk/predis/blob/v1.0/FAQ.md#speaking-about-performances

For example on my synology it is quite a lot easier to install Redis, than it is to try and install the PHP redis extension. This PHP extension can't be installed without a trick.  I assume the same will apply to many users in other setups as well. 
Given that Redis is recommended / mandatory to run "Wh Accumulator" process, it seems like it would be a good idea to make it as easy as possible to get Redis support.

If the emoncms php application could evolve and replace the usage of the PHP-redis extension, I would really appreciate ... and finally use the full potential of Emoncms on Synology.

TrystanLea, 2017-02-12:
What would be involved with using/switching to predis? Would it require changes to redis function names throughout? Sounds useful for new installs but I think it could cause more of a problem than its worth for all the users currently using phpredis. Good to know its an option, if its an easy change we could perhaps add a note about it to the documentation.

yphyph01, 2017-02-13:
Indeed, the usage of some redis function names will have to be updated. So a change is necessary and will impact the current users but the advantage is that ALL users will have the opportunity to use Redis. Currently that's not the case. a part of the users can't use Redis

TrystanLea, 2017-02-13:
Thankyou, could you provide examples of the changes required?

TrystanLea, 2017-02-28:
@glynhudson @pb66 do you have any thoughts on this? I would prefer not to cause any unnecessary support issues and confusion that could result, I will keep using phpredis on emoncms.org but it sounds like there may be advantages to predis on raspberrypi/self installs

pb66, 2017-03-01:
I'm not sure I'm in a position to judge predis and phpredis in a side by side comparison as I'm not that clued up on coding PHP, but based purely on the points raised here I think the effort and changes to a larger user group might out weigh the gains to a much smaller group if "This PHP extension can't be installed without a trick." means it can still be installed with a small inconvenience.

@yphyph01, what does the trick consist of? and what installation methods have you tried? Why is installing phpredis more difficult on synology?

> but the advantage is that ALL users will have the opportunity to use Redis. 

I think it will take more than a change of php extension to ensure that is the case, the main obstacles with using Redis are encountered with shared hosting and with windows, that won't change by changing the extension. To accommodate ALL users we should be looking for an alternative to Redis itself rather than a change of extension.  Many shared-hosting and Windows users are totally excluded from the use Redis, this comes up frequently on the forums.

yphyph01, 2017-03-04:
Hi @pb66 ,
The trick consists of hacking the synology because in this system PHP is delivered as tailored package. Basicly, we can't have access/rights to change it. No way to add a php extension. That's the reason why we can't connect EmonCMS to the Redis server. This could maybe done if you're an expert to hack Synology system.
The advantage to use this predis is for the future: one solution fits all. Emoncms will reduce its external dependancies/constraints but I agree, the current emoncms/redis will be impacted.
Don't forget that predis is NOT a PHP extension. it's a PHP library, that means code included in Emoncms distribution ... independant from server context. It's not replacing extension by another. 
Issues and constraints related to install PHP extension will be removed on Windows, shared server, PI, ....
Regards

TrystanLea, 2017-03-05:
@yphyph01 Ideally it would be possible to chance which redis client is being used via a settings file entry, if the method calls are generally the same $redis->get("key"); etc then this could be possible. 

Have you been able to get emoncms to work with predis? Can you detail the changes made? If the changes are minor it might be easiest to see them in a pull request form, we could then evaluate and make any final changes required if it is a straightforward process.

TrystanLea, 2017-03-08:
@yphyph01 how does predis work on a shared server? and how easy is it to use on windows?
It would actually be a huge step forward in my opinion if we could remove the current requirement for making redis optional.

yphyph01, 2017-03-08:
Hi @TrystanLea 
predis is a PHP "client", a interface to connect to a Redis server. 

The official Redis page (https://redis.io/download ) announces that Redis server is not officially supported on Windows. However some portages exist (see https://github.com/dmajkic/redis/downloads).

Since the predis, is not a PECL, the installation is simple and platform independant. The Predis library (a sampel of PHP files) can be found on Packagist for an easier management of projects dependencies using Composer or on their PEAR channel for a more traditional installation using PEAR. Ultimately, compressed archives of each release are available on GitHub. (please see https://packagist.org/packages/predis/predis)

please see how to install and configure the client in Windows (https://scotch.io/tutorials/getting-started-with-redis-in-php#install-on-windows)
 
Last comment, predis client is officially recommended by Redis (https://redis.io/clients#php)

But, I'll not be able to go further with you, I'm not a developper would be able to go in the technical details of implementation ;-(

TrystanLea, 2019-07-11:
Im closing this as its not something that we have had or likely to have the time resources to implement. The pecl package works for our requirements. If anyone did want to implement a predis port by all means, please do, if the compatibility is close enough so that most redis calls stay the same then we could consider supporting both but its not something we are going to get a chance to look into ourselves unfortunately. Thankyou

closed 2019-07-11 by TrystanLea