emoncms/emoncms #630: Feed names

pm-cz, 2017-01-15:
Hello, in recent version there is a bug which drops many characters from feed names. Is there any reason why the feed names and tags may not contain parentheses, dots, commas and similar ASCII characters?

E.g. I am unable to save a feed name **Rel. humidity [%]** which I would like to use as a multigraph label (that is not directly editable), as the characters will be replaced with preg_replace **Rel humidity** on server. I could create a pull request adding characters permitted e.g. in multigraph name, but I am not sure it that it was not intentional and will not break something else.

glynhudson, 2017-01-15:
Duplicate of https://github.com/emoncms/emoncms/issues/589 ?

I've just responded on that issue

glynhudson, 2017-01-15:
I will close this issue and continue discussion on #589

pm-cz, 2017-01-15:
I don't think this is a duplicate of #589 since it addresses a different issue. 

The _feed_model.php_ already contains unicode support in trunk using `preg_replace('/[^\p{N}\p{L}_\s-:]/u', ...)` in most places. 

But it limits the strings only to unicode letters, whitespaces, _, : and -. Other ASCII letters such as parentheses, percent sign, etc.  are left out (for ' and " this is a must-have feature since the code does not use query builders or escaping and would be prone to SQL injection otherwise).

pm-cz, 2017-01-16:
I wonder, would the pattern `'/[^(-_{-~#-&\p{N}\p{L}\s]/'` be OK? It covers ASCII characters without any quotation mark.

JonMurphy, 2017-01-16:
This may be related to what I saw here:
https://community.openenergymonitor.org/t/unable-to-use-certain-symbols-in-mqtt-topics-with-emoncms-mqtt-input/2753?u=jon

Most UTF8 characters will work with MQTT but it seems only alpha-numeric characters work with emoncms.

TrystanLea, 2017-02-12:
I guess the thing to review here is whether making the pattern more permissive adds any risk of sql injection. It would be good to change all updates and inserts to use prepared queries to mitigate this at the same time, an example of using prepared queries can be found here:

https://github.com/emoncms/emoncms/blob/master/Modules/user/user_model.php#L389

I would be happy to accept pull requests which simultaneously implement both.

emrysr, 2018-06-18:
I am now able to rename my "New Virtual Feeds üaçæêé°C%$£"
can even call it:  ' OR 1=1
cannot use double quotes or tags <h1>Hello WorldÆØÅ!</h1> saves as Hello WorldÆØÅ!

pull request https://github.com/emoncms/emoncms/pull/886