emoncms/emoncms #638: Dashboard is ignoring graph's legend and tag flags

fjriosp, 2017-02-12:
I've configure a graph with multiple bars, a fixed interval, etc. but I don't want to see the legend info so I've unchecked 'legend' and 'feed tag' flags. I've save the graph.

When I add that graph on a dashboard, the legend is shown, but if I go again to the graphs page and open the saved graph, the flag is unchecked and I don't see the legend.

Maybe it's a problem related to the dashboard module, but I don't know.

[image]

[image]

pb66, 2017-02-12:
Hi, See the ["Emoncms Graph module developments"](https://community.openenergymonitor.org/t/emoncms-graph-module-developments/1380?u=pb66) discussion on the OEM forum. It is a known issue with the [emoncms graph module](https://github.com/emoncms/graph).

I have opened an issue for you there too (you may want to subscribe to that issue) 
https://github.com/emoncms/graph/issues/7

Paul-Reed, 2017-02-12:
Fixed by https://github.com/emoncms/graph/commit/91147594012b8fb2b0da121070e3c21eef215bc9

closed 2017-02-12 by Paul-Reed