emoncms/emoncms #640: incompatibility with php-pecl-redis 3.1.1

Ib2cool, 2017-02-19:
It seems data processed by "Wh Accumulator" with package php-pecl-redis 0:3.1.1 are wrong
but correct after downgrade to php-pecl-redis.x86_64 0:2.2.8 correct

with php-pecl-redis 0:3.1.1
input modified to diff for comparison 
[19/Feb/2017:20:20:01 372
[19/Feb/2017:20:21:01 378
[19/Feb/2017:20:22:02 384
[19/Feb/2017:20:23:01 378
[19/Feb/2017:20:24:01 372
[19/Feb/2017:20:25:01 372
[19/Feb/2017:20:26:01 378
[19/Feb/2017:20:27:01 384
[19/Feb/2017:20:28:01 378
[19/Feb/2017:20:29:02 414
Wh Accumulator:
1487532000 Sun, 19 Feb 2017 20:20:00 +0100 180
1487532060 Sun, 19 Feb 2017 20:21:00 +0100 558
1487532120 Sun, 19 Feb 2017 20:22:00 +0100 336.39343261719
1487532180 Sun, 19 Feb 2017 20:23:00 +0100 122.0338973999
1487532240 Sun, 19 Feb 2017 20:24:00 +0100 492
1487532300 Sun, 19 Feb 2017 20:25:00 +0100 264
1487532360 Sun, 19 Feb 2017 20:26:00 +0100 642
1487532420 Sun, 19 Feb 2017 20:27:00 +0100 426
1487532480 Sun, 19 Feb 2017 20:28:00 +0100 204
1487532540 Sun, 19 Feb 2017 20:29:00 +0100 607.86883544922

with php-pecl-redis 0:2.2.8
input modified to diff for comparison 
[19/Feb/2017:21:20:01 432
[19/Feb/2017:21:21:01 432
[19/Feb/2017:21:22:01 426
[19/Feb/2017:21:23:02 426
[19/Feb/2017:21:24:01 408
[19/Feb/2017:21:25:01 408
[19/Feb/2017:21:26:01 408
[19/Feb/2017:21:27:01 408
[19/Feb/2017:21:28:01 414
[19/Feb/2017:21:29:01 414
Wh Accumulator:
1487535600 Sun, 19 Feb 2017 21:20:00 +0100 432
1487535660 Sun, 19 Feb 2017 21:21:00 +0100 432
1487535720 Sun, 19 Feb 2017 21:22:00 +0100 426
1487535780 Sun, 19 Feb 2017 21:23:00 +0100 419.01638793945
1487535840 Sun, 19 Feb 2017 21:24:00 +0100 414.91525268555
1487535900 Sun, 19 Feb 2017 21:25:00 +0100 408
1487535960 Sun, 19 Feb 2017 21:26:00 +0100 408
1487536020 Sun, 19 Feb 2017 21:27:00 +0100 408
1487536080 Sun, 19 Feb 2017 21:28:00 +0100 414
1487536140 Sun, 19 Feb 2017 21:29:00 +0100 414

glynhudson, 2017-02-19:
Thanks a lot for reporting this with such detailed debug into. This bug report is very topical since we have a forum thread discussion ongoing atm regarding the best way to upgrade to php redis 3.1.1. I have posted this issue on the thread:

https://community.openenergymonitor.org/t/php-redis-3-1-1/3124/26?u=glyn.hudson

I wonder what version of php-redis introduced this error? Have you tried 3.0.0? 

The lastest emonSD build currently uses 2.2.7 which works fine and you mention 2.2.8 was also ok.

pb66, 2017-02-21:
@Ib2cool 
Can you explain the data and/or your set up, (preferably on the [OEM forum](https://community.openenergymonitor.org/latest) until confirmed as a bug).

I might be reading the data you provide incorrectly but to me both seem wrong!

pb66, 2017-05-23:
This issue is covered in greater depth in #660  so I'll close this.

closed 2017-05-23 by pb66