emoncms/emoncms #646: Node values missing

bphermansson, 2017-03-01:
I have a custom transmitter that sends values to a local Emoncms installation. It has node number 27. In emonhub.log I see this:
```

19042 From Node : 27
2017-03-01 16:01:16,609 DEBUG    RFM2Pi     19042    Values : [3305, 986.147216796875, 24.0257568359375, 1, 30, 0.2490234375, 31.264968872070312]
2017-03-01 16:01:16,619 DEBUG    RFM2Pi     19042      RSSI : -62
2017-03-01 16:01:16,640 INFO     RFM2Pi     Publishing: emon/emonTXv2/batteryvolt 3305
2017-03-01 16:01:16,656 INFO     RFM2Pi     Publishing: emon/emonTXv2/pres 986.147216797
2017-03-01 16:01:16,674 INFO     RFM2Pi     Publishing: emon/emonTXv2/bmptemp 24.0257568359
2017-03-01 16:01:16,689 INFO     RFM2Pi     Publishing: emon/emonTXv2/uvraw 1
2017-03-01 16:01:16,726 INFO     RFM2Pi     Publishing: emon/emonTXv2/humidity 30
2017-03-01 16:01:16,762 INFO     RFM2Pi     Publishing: emon/emonTXv2/solarV 0.2490234375
2017-03-01 16:01:16,775 INFO     RFM2Pi     Publishing: emon/emonTXv2/power 31.2649688721
2017-03-01 16:01:16,788 INFO     RFM2Pi     Publishing: emon/emonTXv2/rssi -62
2017-03-01 16:01:16,800 INFO     RFM2Pi     Publishing: emonhub/rx/27/values 3305,986.147216797,24.0257568359,1,30,0.2490234375,31.2649688721
2017-03-01 16:01:16,826 INFO     RFM2Pi     Publishing: emonhub/rx/27/rssi -62
2017-03-01 16:01:16,871 DEBUG    RFM2Pi     19042 adding frame to buffer => [1488380476, 27, 3305, 986.147216796875, 24.0257568359375, 1, 30, 0.2490234375, 31.264968872070312, -62]
```

Here I see my seven values plus rssi as expected. 

Then I go to Setup/Inputs and only see this:
`emonTXv2	rssi			4 mins	-62`

In my config I have this:
```
[[27]]
    nodename = emonTXv2
    [[[rx]]]
       names = batteryvolt, pres, bmptemp, uvraw, humidity, solarV, power
       datacodes = h,f,f,h,h,f,f
       scales = 1,1,1,1,1,1,1
       units = V,hPa,C,%,%,V,W
```

Why doesnt all values show up as inputs?

glynhudson, 2017-03-01:
Hi, please post the issue on our active community support forum: https://community.openenergymonitor.org/

I will do my best to help.

closed 2017-03-01 by glynhudson

bphermansson, 2017-04-08:
So I did but I still don't have a solution.
https://community.openenergymonitor.org/t/node-values-missing/3471/4