emoncms/emoncms #647: Database tables for softlinked modules not created

adminde, 2017-03-02:
Modules with only a softlink in the Modules directory do not get their *_schema.php processed correctly. Both at database checkup, as well as in the administrator view.
Whereas the softlinked modules work perfectly fine, if the tables were created already.

Is this a bug or a feature? :)

TrystanLea, 2017-03-02:
@adminde which Module are you trying to symlink? 

I think the backup module (https://github.com/emoncms/backup) is the only module we currently require to be symlinked and this does not have a _schema.php file.

I guess emoncms wasnt really designed for this use case, happy to accept a pull request with required changes.

adminde, 2017-03-08:
Specifically the 4 modules of [this OpenMUC integration](https://github.com/isc-konstanz/emonmuc ) for emoncms :)
I realized that one pretty late and was asking myself if making *_menu.php files aware of links, but *_schema.php files not, is a feature or not.

I just created a (very) small pull request for that.

glynhudson, 2017-03-15:
Merged https://github.com/emoncms/emoncms/commit/2a01a6c51528b0bbc045259f766b557f65228aac

closed 2017-03-15 by glynhudson