emoncms/emoncms #655: Delete old data within a feed

OHarvey, 2017-03-26:
Not sure if this has been previously raised, if it has I apologise. 

I would like the ability to clear data that has been logged in a feed. 

A "clear all" would be very helpful to have in the feeds edit menu. 

Thanks.

Paul-Reed, 2017-03-26:
Agree, with the safeguard of a secondary check - 'This action is irreversible, are you sure that you want to delete this feed's data?' Y/N
This crops up regularly in the forum, and would be a nice addition.

Paul

lxxl, 2017-03-30:
Yep +1 for this.

robertsLando, 2017-10-12:
Any news about this feature? Is there any work in progress? 

I need to delete data of a feed before a date, I also would like to know if there is a way to remove all values of a feed without remove and recreate the feed. I think this feature is a must have in a software like emon, the possibility to delete datas isn't supported in anyway, also an API to delete feeds data would be great for full feeds support.

TrystanLea, 2018-06-27:
@emrysr a couple of pointers to help, we discussed two methods one for the clear all data for a particular feed and the other to delete data before a certain time.

These methods could be implemented in the feed engine's themselves, e.g PHPFina.php: https://github.com/emoncms/emoncms/blob/master/Modules/feed/engine/PHPFina.php

perhaps of the form:

    public function clear($feedid) {
        // clear code...
        // 1. clear feedid.dat
        // 2. set start_time to 0 in feedid.meta, keep interval the same
    }

    public function trim($feedid,$start_time) {
        // trim code...
        // 1. system level trim command? as discussed?
        // 2. change start_time in feedid.meta to new $start_time
    }

thanks a lot!

PR #910 (2018-07-26): Feature/issue 655 clear all old feed data
fixes: #655 
also fixes: #878

this allows for clearing all feed data and "trimming" data up to a given date

[image]
column marked yellow = clear
column marked green = trim

**NOT YET TESTED ON A EMONPI.**
closed 2018-07-26 by TrystanLea