emoncms/emoncms #660: Limitations imposed by emoncms calculations

pb66, 2017-04-19:
I have found emoncms will stop accumulating when a limit defined by the update rate and size of increase reaches a certain level.

In my case recording PV using a power to kWh process to record total kWh's has maxed out at 10000kWh's and has not recoreded any increase since reaching that threshold.

See the ["10000 kWh ceiling in emoncms?"](https://community.openenergymonitor.org/t/10000-kwh-ceiling-in-emoncms/3750?u=pb66) thread on the OEM forum, I can provide more data and screenshots if required.

I'm not sure where to go with this as I'm not good with PHP, this is the ["power to kWh" process in question](https://github.com/emoncms/emoncms/blob/master/Modules/process/process_processlist.php#L272-L300)

I know there was a change, way back to make the "value" field in the feeds table (mySQL) a double but I have no idea if the sums are being done using floats or doubles, also is it possible that Redis could cause a "bottle-neck" somewhere? 

Any assistance appreciated

chaveiro, 2017-04-19:
I remember the cause was adding a very small value to a large float that has lower precision ended up with a close number that when rounded is the same as it started before the add. Are you using redis?

pb66, 2017-04-19:
Yes that is exactly the problem I'm seeing and yes I am using Redis but I do not know enough about how it works to debug. Do you have any suggestions?

joyrider3774, 2017-04-19:
@pb66 do you have a local installation of emoncms ?

if so make backups first (just to be safe) and try this change then update the database:

It's a change i did to my local installation to prevent the problem i had with smaller values and getting the same value after rounding. As soon as i added that a long time ago i never had that problem again (with rounding errors).  Also i don't know what it currently is set at, since i'm not using latest emoncms

[image]

joyrider3774, 2017-04-19:
sorry it probably is not related to this as i read through the old forum thread describing this problem i had initially it seems the problem did not happen with people using redis. https://openenergymonitor.org/forum-archive/node/11774.html . I did not read thourougly enough

TrystanLea, 2017-04-20:
@pb66 I imagine that changing the precision in mysql does not solve your problem as your using redis?

I've been trying to replicate the issue this morning, to no avail so far.
I started by setting up a test in emoncms (master branch, redis enabled, buffered writing enabled) with 2 feeds one starting at 0 kWh the other starting at 9950 kWh. I then setup a power_to_kwh processor on both and pushed the 9950 kWh feed over the 10,000 kWh mark. That seemed to work ok.

For my next step I have tried to isolate the issue down to the bare bones of the power_to_kwh accumulation process.

1. fetch last kwh value from redis
2. calculate the increment in kwh
3. add the increment to the last value
4. save the result in redis

Here is my test script:

    <?php

    $redis = new Redis();
    $connected = $redis->connect("localhost");

    // Initial starting values
    $redis->set("test:kwh1",0);
    $redis->set("test:kwh2",10000);

    $seconds = 0;
    while (true) {

        $power = 1000;
        $timeelapsed = 10;
        $seconds += $timeelapsed;

        $kwh1 = $redis->get("test:kwh1");
        $kwh2 = $redis->get("test:kwh2");

        $kwhinc = ($power * $timeelapsed) / 3600000;

        $kwh1 += $kwhinc;
        $kwh2 += $kwhinc;

        $redis->set("test:kwh1",$kwh1);
        $redis->set("test:kwh2",$kwh2);

        if (($seconds/86400)>=10.0) {
            print "Days: ".number_format($seconds/86400,1)."\n";
            print "kwhinc: ".$kwhinc."\n";
            print "kWh1: ".$kwh1."\n";
            print "kWh2: ".$kwh2."\n";
            die;
        }
    }

Running this script simulating 10 days of accumulation for different power levels: 1W, 10, 100, 1000, 10000, 100000:

    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 2.7777777777778E-6
    kWh1: 0.24000000011927
    kWh2: 10000.2400192
    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 2.7777777777778E-5
    kWh1: 2.4000000011927
    kWh2: 10002.4000192
    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 0.00027777777777778
    kWh1: 24.000000011927
    kWh2: 10024.0000192
    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 0.0027777777777778
    kWh1: 240.00000011927
    kWh2: 10240.0000192
    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 0.027777777777778
    kWh1: 2400.0000011927
    kWh2: 12400.0000192
    trystan@trystan:~/Desktop$ php roundtest.php 
    Days: 10.0
    kwhinc: 0.27777777777778
    kWh1: 24000.000011927
    kWh2: 34000.0000192

Im not really seeing a big difference on the 10,000 kWh mark. 
Starting at 1,000,000 kWh the issue starts to become more visible perhaps?

    Days: 10.0
    kwhinc: 2.7777777777778E-6
    kWh1: 0.24000000011927
    kWh2: 1000000.24192

pb66, 2017-04-20:
Hi Trystan, I just came to post some real data values from the power-to-kwh process having changed the log settings and saw your post, so I haven't digested your tests yet. 

These are small samples for a phptimeseries and a phpfina feed (I keep the phpfina just for use in the apps) as you can see they are the same data and results. 

```
2017-04-20 10:21:06.559|WARN|process_processlist.php|power_to_kwh() feedid=100243 last_kwh=10004.3 kwh_inc=0.0070442777777778 new_kwh=10004.307044278 last_time=1492683660 time_now=1492683665
2017-04-20 10:21:12.233|WARN|process_processlist.php|power_to_kwh() feedid=100243 last_kwh=10004.3 kwh_inc=0.0084551333333333 new_kwh=10004.308455133 last_time=1492683665 time_now=1492683671
2017-04-20 10:21:17.927|WARN|process_processlist.php|power_to_kwh() feedid=100243 last_kwh=10004.3 kwh_inc=0.0056348555555556 new_kwh=10004.305634856 last_time=1492683671 time_now=1492683675
2017-04-20 10:21:23.655|WARN|process_processlist.php|power_to_kwh() feedid=100243 last_kwh=10004.3 kwh_inc=0.007020625 new_kwh=10004.307020625 last_time=1492683675 time_now=1492683680
2017-04-20 10:21:29.170|WARN|process_processlist.php|power_to_kwh() feedid=100243 last_kwh=10004.3 kwh_inc=0.008463 new_kwh=10004.308463 last_time=1492683680 time_now=1492683686
2017-04-20 10:21:40.167|WARN|process_processlist.php|power_to_kwh() feedid=100243  last_kwh=10004.3 kwh_inc=0.0084532666666667 new_kwh=10004.308453267 last_time=1492683690 time_now=1492683696
```

```
2017-04-20 10:21:06.560|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.0070442777777778 new_kwh=10004.307044278 last_time=1492683660 time_now=1492683665
2017-04-20 10:21:12.235|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.0084551333333333 new_kwh=10004.308455133 last_time=1492683665 time_now=1492683671
2017-04-20 10:21:17.928|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.0056348555555556 new_kwh=10004.305634856 last_time=1492683671 time_now=1492683675
2017-04-20 10:21:23.656|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.007020625 new_kwh=10004.307020625 last_time=1492683675 time_now=1492683680
2017-04-20 10:21:29.172|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.008463 new_kwh=10004.308463 last_time=1492683680 time_now=1492683686
2017-04-20 10:21:40.168|WARN|process_processlist.php|power_to_kwh() feedid=100244 last_kwh=10004.3 kwh_inc=0.0084532666666667 new_kwh=10004.308453267 last_time=1492683690 time_now=1492683696
```

The problem seems to be the `last_kwh=10004.3` never increases due to some rounding of the previous new_kwh value as suspected. But I cannot determine where that value comes from, yes I have redis installed but I would expect the value to also be persisted somewhere to survive a restart. 

The "value" field in the mySQL feeds table doesn't seem to alter and the last recorded "value" seems to be 182.38 and last updated on Thu, 18 Aug 2016 14:08:05 GMT. 

That would mean the root/initial value must be coming direct from the phpfina or phptimeseries datafiles directly does it not? if so since they are restricted to 32bit floats by design, would that dictate  the datasize of the Redis field? (I'm guessing a bit here as I do not know how the Redis tables are defined)

I have checked and Redis is still enabled ok.

TrystanLea, 2017-04-20:
Thanks @pb66, the power_to_kwh math runs independently of persistent storage most of the time, unless redis is cleared and then at that point emoncms gets the last value from the phpfina/phptimeseries file on disk.

Which means that the kwh accumulation should run independently of any precision limitations on the persistent storage side, it should even work if the resolution of the storage was 1 kWh as the accumulation math runs using the redis in memory store of the last value...

Your logs suggest that the new value is not being saved to redis somehow... although I imagine the issue will be somewhere else..

do you think you could try checking if there is a redis entry for feed 100244 and what it reads?

    $ redis-cli
    $ hget feed:100244 value

TrystanLea, 2017-04-20:
I wonder if you could also try running my test script above on your system to see if there are any underlying differences in the way php and redis handle floats on your system?

pb66, 2017-04-20:
It looks like you are onto something there

```
pi@emoncmsPi:~ $ redis-cli
127.0.0.1:6379> hget feed:100244 value
(nil)
127.0.0.1:6379> hget feed:100243 value
(nil)
```

TrystanLea, 2017-04-20:
Great, do you see any entries in redis using:

    keys *

pb66, 2017-04-20:
Yes I have 3848 keys listed, that tallies with the admin page in emoncms too.
here is the last 10, the rest of the list looks much the same
```
3839) "emoncms:feed:100912"
3840) "emoncms:input:lastvalue:100248"
3841) "emoncms:input:100919"
3842) "emoncms:feed:101055"
3843) "emoncms:input:lastvalue:101137"
3844) "emoncms:feed:101017"
3845) "emoncms:input:100950"
3846) "emoncms:input:151"
3847) "emoncms:feed:100281"
3848) "emoncms:feed:101230"
```

TrystanLea, 2017-04-20:
ah ok, your using the redis prefix. Could you try:

hget emoncms:feed:100244 value
hget emoncms:feed:100244 time

pb66, 2017-04-20:
```
pi@emoncmsPi:~ $ redis-cli
127.0.0.1:6379> hget emoncms:feed:100244 value
"10004.5"
127.0.0.1:6379> hget emoncms:feed:100244 time
"1492688770"
127.0.0.1:6379>
```

TrystanLea, 2017-04-20:
Ok that appears to be saving ok, although the precision of value is very low and similar to what your log file suggested. 

It should look more like:

hget feed:3 value
"40.232192993164"

pb66, 2017-04-20:
Running your script (mod'ed slightly to accept the power via command line and print the power to id the test) at 0.001 through to 10000000 gave these results
```
pi@emoncmsPi:~ $ php ./test_redis.php 0.001
Test Power: 0.001w
Days: 10.0
kwhinc: 2.7777777777778E-9
kWh1: 0.00025197877777778
kWh2: 10000.000000003

pi@emoncmsPi:~ $ php ./test_redis.php 0.01
Test Power: 0.01w
Days: 10.0
kwhinc: 2.7777777777778E-8
kWh1: 0.0025197877777778
kWh2: 10000.000000028

pi@emoncmsPi:~ $ php ./test_redis.php 0.1
Test Power: 0.1w
Days: 10.0
kwhinc: 2.7777777777778E-7
kWh1: 0.025197877777778
kWh2: 10000.000000278

pi@emoncmsPi:~ $ php ./test_redis.php 1
Test Power: 1w
Days: 10.0
kwhinc: 2.7777777777778E-6
kWh1: 0.25197877777778
kWh2: 10000.000002778

pi@emoncmsPi:~ $ php ./test_redis.php 10
Test Power: 10w
Days: 10.0
kwhinc: 2.7777777777778E-5
kWh1: 2.5197877777778
kWh2: 10000.000027778

pi@emoncmsPi:~ $ php ./test_redis.php 100
Test Power: 100w
Days: 10.0
kwhinc: 0.00027777777777778
kWh1: 25.197877777778
kWh2: 10000.000277778

pi@emoncmsPi:~ $ php ./test_redis.php 1000
Test Power: 1000w
Days: 10.0
kwhinc: 0.0027777777777778
kWh1: 251.97877777778
kWh2: 10000.002777778

pi@emoncmsPi:~ $ php ./test_redis.php 10000
Test Power: 10000w
Days: 10.0
kwhinc: 0.027777777777778
kWh1: 2519.7877777778
kWh2: 10000.027777778

pi@emoncmsPi:~ $ php ./test_redis.php 100000
Test Power: 100000w
Days: 10.0
kwhinc: 0.27777777777778
kWh1: 25197.877777778
kWh2: 35919.977777778

pi@emoncmsPi:~ $ php ./test_redis.php 1000000
Test Power: 1000000w
Days: 10.0
kwhinc: 2.7777777777778
kWh1: 251978.77777778
kWh2: 262770.77777778

pi@emoncmsPi:~ $ php ./test_redis.php 10000000
Test Power: 10000000w
Days: 10.0
kwhinc: 27.777777777778
kWh1: 2519787.7777778
kWh2: 2530587.7777778
```

Looking at these numbers and given the rounding to 2 decimals we're seeing it looks like it would need a power of ~2000w over the 10s period to increment (round up), since I am actually updating at 5s intervals it would probably need ~4000w power. However it wasn't incrementing at over 5500w the other day so I wonder if it's possible it's being truncated rather than rounded?

pb66, 2017-04-20:
Also check the precision of some other calculated feeds and found they are all pretty short

```
pi@emoncmsPi:~ $ redis-cli
127.0.0.1:6379> hget emoncms:feed:100467 value
"1081.88"
127.0.0.1:6379> hget emoncms:feed:100239 value
"5301.34"
127.0.0.1:6379> hget emoncms:feed:100249 value
"3599.97"
127.0.0.1:6379> hget emoncms:feed:101808 value
"257.897"
127.0.0.1:6379> hget emoncms:feed:100243 value
"10004.7"
127.0.0.1:6379> hget emoncms:feed:100239 value
"5301.4"
127.0.0.1:6379> hget emoncms:feed:100467 value
"257.975"
127.0.0.1:6379> hget emoncms:feed:100467 value
"1081.94"
127.0.0.1:6379> hget emoncms:feed:100467 value
"1081.95"
127.0.0.1:6379> hget emoncms:feed:100467 value
"1081.95"
127.0.0.1:6379> hget emoncms:feed:100467 value
"1081.95"
```

TrystanLea, 2017-04-20:
That's interesting! Could you try setting a redis entry manually via command line:

    hmset test:feed:1000 value 200.242352

then

    hmget test:feed:1000 value
    "200.242352"

If that returns a rounded number, maybe its a redis version or php redis client version issue?

pb66, 2017-04-20:
No, that test seems to be fine. I even tried to overload it with a couple of extra zeros.

```
pi@emoncmsPi:~ $ redis-cli
127.0.0.1:6379> hmset test:feed:1000 value 200.242352
OK
127.0.0.1:6379> hmget test:feed:1000 value
1) "200.242352"
127.0.0.1:6379> hmset test:feed:1000 value 200.2423520000000000000000000000000000000000000000000001
OK
127.0.0.1:6379> hmget test:feed:1000 value
1) "200.2423520000000000000000000000000000000000000000000001"
127.0.0.1:6379>
```

TrystanLea, 2017-04-20:
Slowly narrowing it down.. hopefully.. could you test:

<?php
$redis = new Redis();
$connected = $redis->connect("localhost");
$redis->set("test:kwh1",200.24235200000023);
print $redis->get("test:kwh1");

pb66, 2017-04-20:
```
pi@emoncmsPi:~ $ php ./test_redis1.php
200.242
```
That looks like it is replicating the rounding issue albeit to 3 places not 2, probably due to a smaller value so I tried it with the new_kwh=10004.307044278 from the first log sample I posted
```
 pi@emoncmsPi:~ $ php ./test_redis1.php
10004.3
```
Spot on...

So what does that suggest?

TrystanLea, 2017-04-20:
Great, well, in trying to get my redis client version by seeing if there was a new version of redis available (by running sudo pecl install redis) I have now upgraded this stock emonpi sd and am now getting the same results as you! Which confirms that its being caused by a change in the redis-client.
I now have version redis-3.1.2
Im trying to find out now what has changed and whether the solution is to roll back or implement a fix elsewhere. If the number is converted to a string with:

    $redis->set("test:kwh1",(string)10004.307044278);

it works

TrystanLea, 2017-04-20:
testing with redis-3-1.0 results in the same issue.

Im running version redis-3.0.0 on my laptop but it wouldn’t install on the pi as php version was not recent enough:

    "pecl/redis requires PHP (version >= 7.0.0, version <= 7.1.0, excluded versions: 6.0.0), installed version is 5.6.27-0+deb8u1"

reverting to version redis-2.2.8 on the pi works

    sudo pecl uninstall redis
    sudo pecl install redis-2.2.8

There must be a setting somewhere to tell phpredis how to convert the number.

There is a note in the change log here https://pecl.php.net/package/redis/3.1.1RC2 suggesting the related modification:

    * Fix double to string conversion function (@yatsukhnenko)

JonMurphy, 2017-04-20:
I'm running redis 2.8.17:

```
pi@emonpi(ro):myApps$ redis-cli
127.0.0.1:6379> info server
# Server
redis_version:2.8.17
redis_git_sha1:00000000
redis_g...
```
and it looked like Paul was running the [same](https://community.openenergymonitor.org/t/10000-kwh-ceiling-in-emoncms/3750/2?u=jon)

But I get a different answer then Paul:
```
pi@emonpi(ro):myApps$ php redisTest.php
200.242352
```

pb66, 2017-04-20:
Yep it looks like phpredis.

I have rolled back to v2.2.8 on the emonPi and modded your last test script to
```
<?php
$redis = new Redis();
$connected = $redis->connect("localhost");
print "\n";

$redis->set("test:kwh1",10004.307044278);
print '$redis->set("test:kwh1",10004.307044278);';
print "\n";
print $redis->get("test:kwh1");
print "\n\n";

$redis->set("test:kwh1","10004.307044278");
print '$redis->set("test:kwh1","10004.307044278");';
print "\n";
print $redis->get("test:kwh1");
print "\n\n";
```
which returns 
```
pi@emoncmsPi:~ $ php ./test_redis1.php

$redis->set("test:kwh1",10004.307044278);
10004.307044278

$redis->set("test:kwh1","10004.307044278");
10004.307044278

```
on the emonPi (v2.2.8) and
```
pi@emoncmsPi:~ $ php ./test_redis1.php

$redis->set("test:kwh1",10004.307044278);
10004.3

$redis->set("test:kwh1","10004.307044278");
10004.307044278
```
on my main server (v3.1.2)

pb66, 2017-04-20:
To roll back in one line you can use `sudo pecl install -f redis-2.2.8` the -f forces the rollback without having to uninstall.

In the v3.1.1 changelog there is a entry stating "* ARM rounding issue in tests (@remicollet)" so there maybe value in raising this as an issue over on phpredis https://github.com/phpredis/phpredis, they maybe working on this too.

pb66, 2017-04-20:
@JonMurphy "and it looked like Paul was running the same" the emoncms admin page only shows the version of redis itself. "phpredis" (just called "redis" in pecl) is the php interface to redis, a seperate PHP Module installed to break out Redis functions within PHP.

TrystanLea, 2017-04-20:
If I upgrade my laptop version of phpredis to 3.1.2 my apache2 server crashes :( so I rolled it back to 3.0.0.

pb66, 2017-04-20:
What version of PHP is the laptop running? v3 is supposed to be php v7 only. v3.1 was the first version to serve both php5 and php7

TrystanLea, 2017-04-20:
I've raised it as an issue on the phpredis github https://github.com/phpredis/phpredis/issues/1156

pb66, 2017-04-20:
Doh! so did I https://github.com/phpredis/phpredis/issues/1157

TrystanLea, 2017-04-20:
:) looks like best solution for now is to roll back to phpredis-2.2.8 or potentially try php7 with a newer version, I cant seem to get phpredis-3.1.2 running on my laptop with php7 but maybe I need to try upgrading php further. Interested to hear of others experience of running phpredis-3.1.0+ on php7

pb66, 2017-04-20:
Yeah, I sorta guessed that and have rolled it back for now,.

Those 2 feeds are now way off what they should be, is it possible to rebuild them using the [power_to_kwh.php](https://github.com/emoncms/usefulscripts/blob/master/process/power_to_kwh.php) script? The power feed is phptimeseries and has datapoints every 5s

JonMurphy, 2017-04-20:
I see the phpredis version number here:
http://emonpi/emoncms/php-info.php#module_redis

Would it be OK (or worthwhile?) to add it here to aid in debugging?

[image]

```
pi@emonpi(ro):myApps$ php -r 'phpinfo();' | grep Redis
Redis Support => enabled
Redis Version => 2.2.7
pi@emonpi(ro):myApps$ 
```

joyrider3774, 2017-04-20:
hey john if it's ok to add it and you do it eventually. Please think about windows users as grep does not exist there.. What i'm saying is don't execute the command's if windows is detected, that file should already have other exclusions dealing with linux specific commands & windows (i had problems with it in the past, i run emoncms (only) on windows)

JonMurphy, 2017-04-20:
got it!  

Someone smarter than me added `if (strtoupper(substr(PHP_OS, 0, 3)) === 'WIN') { echo "n/a"; }`

to some code I had helped add `if (file_exists('/usr/sbin/mosquitto')) { echo exec('/usr/sbin/mosquitto -h | grep -oP \'(?<=mosquitto\sversion\s)[0-9.]+(?=\s*\(build)\''); } } `

so I'll give that a try!   Jon

glynhudson, 2017-04-20:
Nice work guys trying to debug this. I've noticed that @TrystanLea  has had a reply on https://github.com/phpredis/phpredis/issues/1156

A fix has just been committed:
 
https://github.com/phpredis/phpredis/commit/4e56ba804cdfcaae36b98f1bbdcf9ffb4e77b6e2

pb66, 2017-04-21:
@JonMurphy 
In PHP the version of a php extension module can be returned using something like 
```
$version = phpversion("redis");
```
this will also work on windows, (not that windows users have any interest in this particular module). 

Do you think we should single out the (php)redis module to show a version, there are many modules as you see in the last section of your admin page screenshot above.

Maybe changing that section by editing [this line](https://github.com/emoncms/emoncms/blob/master/Modules/admin/admin_main_view.php#L331) from 
```
<tr><td class="subinfo"></td><td>Modules</td><td colspan="2"><?php while (list($key, $val) = each($system['php_modules'])) { echo "$val &nbsp; "; } ?></td></tr>
```
to
```
<tr><td class="subinfo"></td><td>Modules</td><td colspan="2"><?php while (list($key, $val) = each($system['php_modules'])) { echo "$val(".phpversion($val).") &nbsp; "; } ?></td></tr>
```
would prove easier and more useful for debugging in the future as it adds the version details of each module (where available) in one go.

[image]

TrystanLea, 2017-04-21:
@pb66 yes you can rebuild the feed with the usefulscripts/process/power_to_kwh.php script.

It will ask you if you would like to create a new feed which I would recommend, then when it asks for interval set to 10 and then max power level to whatever is suitable e.g 24000

TrystanLea, 2017-04-21:
I've made a couple of edits to process/power_to_kwh.php so that it reloads the last value correctly and provides a clearer way to skip the max power level option and interval change option.

JonMurphy, 2017-04-21:
@pb66 - I like what you added for displaying the version number!  Can you add the version changes to the master/stable code?

The confusion for me was the redis versus phpredis references.  Thank you for explaining!  I am trying to understand the redis role within emon.  Following along with the above tests & comments helped.  Hope I didn't get in the way!

pb66, 2017-04-21:
@JonMurphy, I've submitted a PR for review, see what happens.

@TrystanLea, I can't see those edits, have you pushed them to the repo yet?

Is the usefulscripts/process/power_to_kwh.php script able to rebuild the phptimeseries feed too or just the phpfina?

therealchalz, 2017-04-24:
Disclaimer: I know nothing about emon/emoncms, redis, or any of the specific technologies you are talking about here.  I also know every little about PHP, which seems to be what is doing the accumulation arithmetic.  

But I do know a bit amount about numerical analysis and floating point calculations.  

From what I can gather from taking a quick peek around, the accumulation function will always eventually fail, or at least will always have cases where it will not give the correct mathematical result, because (I believe) that the PHP code always works on floats (or doubles, as the case may be).  In particular, `$new_kwh = $last_kwh + $kwh_inc;` will not yield mathematically acceptable results once $last_kwh is sufficiently larger than $kwh_inc.  Arithmetic on floating point values of vastly different orders will give you trouble unless your arithmetic process accounts for the error; the most popular algorithm for correcting this is [Kahan's algorithm](https://en.wikipedia.org/wiki/Kahan_summation_algorithm).  This works because the arithmetic operations are between floating point values of the same order, and something like this is required if you want to sum up and series of floating point values of potentially unbounded length (like power consumption, for example).

Of course I may have misunderstood how your system works or this might not be related, but in any case you should familiarize yourselves with such floating point arithmetic errors and solutions because you don't want to fall into the traps (and because math is fun). :)

JonMurphy, 2017-04-24:
this is fun!  So what decides when $last_kwh is sufficiently larger?  Or how is the first "c" determined?

therealchalz, 2017-04-25:
> So what decides when $last_kwh is sufficiently larger?

That depends on what error you find acceptable.  

Floating point numbers have a value of (sign) * 2^(exponent) * mantissa.  The difference between a float and a double is simply the number of bits available for (exponent) and (mantissa).  

We can simplify things a bit by looking at a decimal analogue, where we say value = 10^(exponent) * mantissa (aka scientific notation, essentially) where we have a finite number of values we can use for exponent and mantissa, and where mantissa is a value between 1 and 10 (with a finite number of decimal places).  If you limit yourself to say, 2 decimal places for simplicity, and start to play around, you can find some equations that work out well, like `(10^0 * 3.00) + (10^1 * 3.00) = (10^1 * 3.30)` and others that don't work out quite so well, like `(10^0 * 3.01) + (10^1 * 3.01) ~ (10^1 * 3.31)`.  As the order of the numbers (or the exponents) tend further and further apart, fewer of the significant digits of the smaller number make it into the resulting sum.  By now you can probably see if the numbers are far enough apart that you can keep summing over and over and over and the result will never change (although long before you get to that point you start losing precision).

> how is the first "c" determined?

You just start with 0.  In this case I figure on initial setup/installation you set $kwh to 0, but you also have to track the error, and have some other value $kwh_error = 0.  Then every time you increment you'd have to pull those from the DB, update as appropriate, and then resave them both.

JonMurphy, 2017-04-25:
got it!  Thank you!  I grabbed the pseudocode and gave it a try within javascript.  But I haven't gotten to the point of  understanding it or seeing how the Kahan's algorithm helps.

pb66, 2017-05-16:
Still waiting for the fix to reach the pecl repo. https://github.com/phpredis/phpredis/issues/1175 but it's gone a bit quiet

spike3312, 2017-05-23:
Hi Guys,

Looks like I've just hit this wall as well, my system has been logging several power usages at home and working great until my usage has dropped off when i went on holiday, at which point my KWh/d has been accumulating 6.x KWh/d and my total KWh readings have only been increasing by 0.1KWh per day

I'm running Emoncms V9.8.2, Redis 2.8.17

pb66, 2017-05-23:
@spike3312 Do you know what version of PHP you are running?

If you are running v5.x (>7) you can over come this by using PHP redis v2.2.8, I assume you've installed using PECL if you currently have v3.1.2, if that is the case you should be able to rollback by simply running

> sudo pecl install -f redis-2.2.8

There should be a fix in v3.1.3, which shouldn't be _too_ long away. I've already asked over at https://github.com/phpredis/phpredis/issues/1175 when that might be, maybe another gentle nudge is needed?.

spike3312, 2017-05-24:
Hi @pb66 I am running PHP Version 5.6.30-0+deb8u1.

I found that command to roll back to 2.2.8 in another thread and ran it. It has helped a lot my total KWh has climbed by 3.4 so far since midnight, but my separate KWh/d feed is showing 5.33 which is quite different.

pb66, 2017-05-24:
What timezone have you selected in emoncms and what timezone are you in? 

If I recall correctly kWh/d "daily feeds" are fixed to UTC so there is always likely to be a discrepancy, however 1.9kWh is quite a difference unless you are several timezones away from UTC.

Can you make a note of the kWh and kWh/d feeds and then recheck an hour or too later to see if the increase on each feed is similar? or check the kWh/d is zero at midnight?

Are the 2 processes (power to kWh and power to kWh/d) directly next to each other in the input processing (ie no other possible influences)  

What sort of magnatude are the values in your kWh feed? as explained above this is likely to occur at some point with any use of floats and doubles, using doubles extends the threshold significantly (maybe from 7 digits to 14 digits of accuracy) but it doesn't prevent it all together.

spike3312, 2017-05-24:
I have selected Europe/London (GMT+1), I am about an hours drive from London so within this timezone.

That would make sense with the time difference, however i am averaging about a 300W load, which shouldn't put the values that far apart.

I've made a note of the two values and will check back later to see how they change.

They are indeed on top of each other in the process list.

My total kWh feed is 6018.37.

After dinner and an hour later my total kWh feed has risen by 0.322 and the kWh/d feed has risen by 0.33, which is a rather close relationship.

Would moving to two long integers not be a better idea, one as kWh's and one as decimals, this should alleviate any rounding errors.

I've put a python script together which will log the kWh/d feed at midnight so i can see what it is tomorrow without having to stay up until then and will report back tomorrow

spike3312, 2017-05-25:
My script logged a value of 0.000525525 on the kWh/d feed at midnight, pretty close to zero.

Must have been a one off yesterday having rolled back the PHP redis version late the night before, so far my kWh feed has risen by 2.564 since midnight and my kWh/d feed is showing 2.56, and therefore right back on track!

Will keep an eye on it for a few days see how it goes.

Many thanks for the help.

glynhudson, 2017-10-24:
Any updates on this? Is the current recommendation to stick with redis 2.2.8?

pb66, 2017-10-27:
There has been quite a bit of activity with phpredis recently. I am currently using version 3.1.4 on a live RPi emoncms server running Raspian Jessie (php5) and on a live Debian8 Jessie (php5) hosted VM server, both seem to be behaving.

I'm now confident with using PECL to install the latest version, the guys over at phpredis seem to be keeping that well up to date, the debian packages are currently 2.2.5 for Jessie and 3.1.1 for Stretch, both of these have issues (hence we use 2.2.8 and this thread about >3 versions). IMO as long as we stick to using PECL for the time being it should be safe to not specify a version and possibly include a "PECL upgrade redis" line in the emonSD updater whilst the Apache server is off line (should always restart apache after updating phpredis, perhaps a "cli-redis flush all" too?

There is a user selection in the install/upgrade process and I have been using this command for a non-interactive install/upgrade `sudo pecl install redis <<<'' `. But please do your own tests too as I have not been testing on a emonPi or emonSD.

If you would prefer to stick with specifying a version in your installer until you test the next incarnation you could use `sudo pecl install redis-3.1.4 <<<'' ` for example. 

Also remember you need to remove the git and deb versions if you roll out an update to your current emonSD image.

closed 2018-07-10 by TrystanLea