emoncms/emoncms #673: Bargraph tooltip blocks buttons

Paul-Reed, 2017-05-12:
I don't personally use the bargraph, but just noticed that when a bar is selected, it's value & date appear in the graph frame top-left, on top of the D, W, M, & Y buttons, meaning that they cannot be selected.
If the graph size is really wide, then there would be sufficient space, but on smaller/medium graphs this presents a problem.
A possible solution may be to ensure that the label disappears on 'mouseout'.
Currently the label appears on 'mouseover' of the respective bar, but is then retained no matter where the mouse is moved.

[image]

firefox7518, 2017-05-17:
Yes, I noticed this as well. 
From my point of view they should be outside the graph or a button on the right top corner should show them when needed and then they are fully visible not just opaque.

I solved this for my instances manually by moving them up. Of course this way i will loose a little bit of graph area but it was more annoying not be able to use the buttons sometimes or the graph.
[image]

glynhudson, 2017-10-05:
I think this issue better located on the dashboard repo: https://github.com/emoncms/dashboard/issues

emrysr, 2018-07-12:
This is actually an issue fixed in the vis module of emoncms... 
I've closed the issue on emoncms/dashboard repo (https://github.com/emoncms/dashboard/issues/163)

TrystanLea, 2018-07-15:
I wonder if it would be possible to place the info below the buttons, and right-aligned but only when there is not enough room to fit it inline, what do you think @emrysr

emrysr, 2018-07-23:
@TrystanLea - The buttons are only right aligned in the larger view -(screenshot below) 

It will take a bit of work to create the responsive css required to handle dropping the tooltip for the different iframe widths. eg.

[image]

I think the bottom position suits all the variations without jumping about in the different sizes.

PR #919 (2018-07-12): moved bargraph value hover container to centre bottom
fixes #673 

TEXT WAS OVERLAYING BUTTONS
[image]

FIXED BUY MOVING TEXT
[image]
closed 2018-07-31 by emrysr