emoncms/emoncms #677: no inputs to emoncms

nigeldodd, 2017-05-20:
I have just installed the emonSD-07Nov16 image on a system that had previously worked some time ago. I have rebooted the Arduino in the garage with the sensors attached, I have rebooted the pi several times. 

I am somewhat defeated by the complexity of this new software. Are there instructions somewhere to test that the Pi is receiving data? How can I proceed to identify and fix the problem? The lights on the Arduino box and the pi seem to be flashing every 10 seconds or so: does this mean successful communication between the two?
Here is the output from ...../emoncms/config/view

2017-05-20 08:32:49,419 INFO     MQTT       connection status: Connection successful
2017-05-20 08:32:49,450 DEBUG    MQTT       CONACK => Return code: 0
2017-05-20 08:32:49,596 INFO     MQTT       on_subscribe
2017-05-20 08:47:22,390 DEBUG    MainThread SIGINT received.
2017-05-20 08:47:22,393 INFO     MainThread Exiting hub...
2017-05-20 08:47:22,480 INFO     MainThread Exit completed
2017-05-20 08:47:26,038 INFO     MainThread EmonHub emonHub 'emon-pi' variant v1.2
2017-05-20 08:47:26,041 INFO     MainThread Opening hub...
2017-05-20 08:47:26,044 INFO     MainThread Logging level set to DEBUG
2017-05-20 08:47:26,047 INFO     MainThread Creating EmonHubJeeInterfacer 'RFM2Pi' 
2017-05-20 08:47:26,057 DEBUG    MainThread Opening serial port: /dev/ttyAMA0 @ 38400 bits/s
2017-05-20 08:47:28,091 WARNING  MainThread Device communication error - check settings
2017-05-20 08:47:28,094 INFO     MainThread Setting RFM2Pi frequency: 433 (4b)
2017-05-20 08:47:29,099 INFO     MainThread Setting RFM2Pi group: 210 (210g)
2017-05-20 08:47:30,103 INFO     MainThread Setting RFM2Pi quiet: 0 (0q)
2017-05-20 08:47:31,107 INFO     MainThread Setting RFM2Pi baseid: 5 (5i)
2017-05-20 08:47:32,112 INFO     MainThread Setting RFM2Pi calibration: 230V (1p)
2017-05-20 08:47:33,116 DEBUG    MainThread Setting RFM2Pi subchannels: ['ToRFM12']
2017-05-20 08:47:33,120 DEBUG    MainThread Interfacer: Subscribed to channel' : ToRFM12
2017-05-20 08:47:33,123 DEBUG    MainThread Setting RFM2Pi pubchannels: ['ToEmonCMS']
2017-05-20 08:47:33,126 DEBUG    MainThread Interfacer: Subscribed to channel' : ToRFM12
2017-05-20 08:47:33,131 INFO     MainThread Creating EmonHubMqttInterfacer 'MQTT' 
2017-05-20 08:47:33,136 INFO     MainThread MQTT Init mqtt_host=127.0.0.1 mqtt_port=1883 mqtt_user=emonpi
2017-05-20 08:47:33,146 DEBUG    MainThread MQTT Subscribed to channel' : ToEmonCMS
2017-05-20 08:47:33,150 INFO     MainThread Creating EmonHubEmoncmsHTTPInterfacer 'emoncmsorg' 
2017-05-20 08:47:33,156 DEBUG    MainThread emoncmsorg Subscribed to channel' : ToEmonCMS
2017-05-20 08:47:33,259 INFO     MQTT       Connecting to MQTT Server
2017-05-20 08:47:33,278 INFO     MQTT       connection status: Connection successful
2017-05-20 08:47:33,292 DEBUG    MQTT       CONACK => Return code: 0
2017-05-20 08:47:33,398 INFO     MQTT       on_subscribe

The only thing that stands out to me is 
2017-05-20 08:47:28,091 WARNING  MainThread Device communication error - check settings
but how do I check the settings? Which settings?

I'm sure I am missing a key piece of information about how to proceed under circumstances like this but I can't find any relevant documentation.

Paul-Reed, 2017-05-20:
Hi Nigel, have you asked for help in the openenergymonitor community forum?
Git issues are really for logging and dealing with confirmed issues that require attention, and you may find that you will get a better response if you post this in the forum as it is of a general nature, rather than specific.

Paul

closed 2017-05-20 by Paul-Reed

nigeldodd, 2017-05-20:
I am sure you are right. I had not understood the distinction.