emoncms/emoncms #679: Set "max age" to feed data to clean up database from old data

robertsLando, 2017-06-07:
I'm looking for a way to easly delete old feed data but I can't find it, there isn't any API that allows to delete Feed data older than a specific date. I think it would be great to set a "max age" property to all feeds that allows to keep database "clean" and never get the HD/SSD memory full.

Paul-Reed, 2017-06-08:
Not currently, but it's a great idea.
This crops up regularly in the forum, especially by new users who experiment in setting up feed processes, then want to delete the initial feed data.

Paul

robertsLando, 2017-06-12:
After many tries I have found this solution but it just works with MySql Engine by using MySql events:

```
DELIMITER |

CREATE EVENT AutoDeleteOldFeedData
ON SCHEDULE EVERY 1 DAY
ON COMPLETION PRESERVE
DO BEGIN
SET @querys = (SELECT CONCAT('DELETE FROM ' ,  table_name , ' WHERE time < CURRENT_DATE - INTERVAL 1 YEAR;')
				FROM information_schema.tables
				WHERE table_schema = 'emoncms'
				AND table_name LIKE 'feed__%');

PREPARE stmt FROM @querys;
EXECUTE stmt;
DEALLOCATE PREPARE stmt;
END |

DELIMITER ;
```

Some utils comands:

```
SET GLOBAL event_scheduler = ON; //ENABLE EVENTS DAEMON

SHOW PROCESSLIST; //CHECK IF EVENTS ARE ENABLED (CHECK IF THERE IS EVENT ESCHEDULER DAEMON IN LIST)

SHOW EVENTS; //SHOWS THE LIST OF EVENTS

DROP EVENT AutoDeleteOldFeedData; //DELETE THE EVENT

SHOW CREATE EVENT AutoDeleteOldFeedData; //CHECK EVENT CREATE CODE

```

emrysr, 2018-07-25:
Hi @robertsLando,

Thanks for your mysql events approach. I am adding a  the clear() and trim() functions to the mysql engine in pull request: https://github.com/emoncms/emoncms/pull/910

I've also added 2 new api endpoints not yet in the documentation - they are not yet implemented for all engines.
- /feed/clear.json?id=0
- /feed/trim.json?id=0&start_time=0

currently only for **mysql** and **phpfina** engines.

Calling the trim() api endpoint using a cronjob should allow you to schedule clearing out old data - you'll have to calculate the new start_time timestamp as it doesn't currently work with relative time parameters

TrystanLea, 2018-08-01:
This is now implemented as a manual option that a user can click on periodically https://github.com/emoncms/emoncms/pull/946 thanks to @emrysr

closed 2018-08-22 by TrystanLea