emoncms/emoncms #682: Authorization Headers are not passed to EmonCMS under FastCGI Apache

davidski, 2017-06-20:
Reference: https://community.openenergymonitor.org/t/using-authorization-header-for-emoncms-api-access/4358

EmonCMS as deployed on the emonSD image does not received `Authorization: Bearer FOO` headers in the `$_SERVER` array. This causes use of auth headers to be ineffective in standard local deployments. Inserting the following block into the `.htaccess` file allows API auth to work as expected in this scenario:

```
# Rewrite auth header
# https://cweiske.de/tagebuch/php-apache-authorization.htm
<IfModule mod_setenvif.c>
  SetEnvIf Authorization "(.*)" HTTP_AUTHORIZATION=$1
</IfModule>
```

chaveiro, 2017-06-21:
Fix here https://github.com/emoncms/emoncms/pull/683

glynhudson, 2017-06-21:
Thanks, merged :+1:

closed 2017-06-21 by glynhudson

davidski, 2017-06-28:
@glynhudson Has this been deployed to emoncms.org yet? Authorization headers seem to be ignored there right now. 😢

pb66, 2017-06-28:
It's unlikely, emoncms.org trails behind the main emoncms repo by some time and not everything gets implemented on there. You would be better off raising this issue on the OEM forum or failing that maybe even over on the [emoncms/emoncmsorg](https://github.com/emoncms/emoncmsorg) repo.