emoncms/emoncms #684: Input Delete API Documentation Incorrect

davidski, 2017-06-28:
The API docs for the input/delete API specify a parameter of `id=<ID>`. The actual parameter needs to be `inputid=<ID>`. Verified against the hosted emoncms.org documentation and API.

Reference https://github.com/emoncms/emoncms/blob/master/Modules/feed/Views/feedapi_view.php#L105

pb66, 2017-06-28:
The input/delete api is indeed `inputid=` as documented in the "input api help" page (ref https://github.com/emoncms/emoncms/blob/master/Modules/input/Views/input_api.php#L77). The link you provide is to the "feed api help" page, which shows the feed/delete api as using `id=`.  

Links to the emoncms.org api documentation pages

https://emoncms.org/site/api#input

https://emoncms.org/site/api#feed

Ahh! I think I've spotted what you might actually be meaning to report, the emoncms.org input/delete docs say `id=` not `inputid=` (ref https://github.com/emoncms/emoncmsorg/blob/master/emoncms/Modules/input/Views/input_api.php#L56), is that what you are referring to? The emoncms.org variant of emoncms has it's own repo at [emoncms/emoncmsorg](https://github.com/emoncms/emoncmsorg). I've raised an issue there pointing here..

davidski, 2017-06-28:
Aye. I did not realize that emoncms and emoncms.org had separate repos. 😕

closed 2017-10-05 by glynhudson