emoncms/emoncms #728: Revise install default to Raspbian Stretch

Paul-Reed, 2017-09-29:
The current raspberry pi operating system is now 'Raspbian Stretch' which is based upon Debian 9, and without resorting to raspberry archives, 'Jessie' has been removed as a install option from raspberrypi.org/downloads

In the emoncms docs repo, there is now a [build guide for stretch](https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/stretch.md) which updates a number of dependencies, and also introduces open-source Mariadb as a successor to MYSQL.

A number of successful installations have now been completed using the guide, however the default option presented to users is still the Jessie version.

@glynhudson @TrystanLea - whenever you feel the times is right to set the default guide to 'Stretch', please update this issue, and I'll make the necessary changes to the links & file names in the build guide to reflect the change.

Paul

glynhudson, 2017-10-02:
Yes, I think it would be a good idea to make Stretch the recommended default for new systems. 

From what I've read it seems that Jessie can be successfully updated to get the latest security updates via the package manger, however we will look at updating our pre-built SD card image in due course. Please feedback any issues using Stretch. 

Thanks again to everyone for your help.

Paul-Reed, 2017-10-04:
Updated by https://github.com/emoncms/emoncms/pull/729

closed 2017-10-04 by Paul-Reed