emoncms/emoncms #744: Enhanced input processing display labels

pb66, 2017-11-17:
Here is a small example of an identical set of inputs on emoncms.org and locally hosted emoncms (only the input name differ)

[image]

[image]

The emoncms.org labels are IMO more informative, compact, colorful and largely language agnostic, I would like to see a return to those tags, but I've been reluctant to implement local changes that need stashing every time I upgrade and may cause merge conflicts etc.

Would there any interest (or objections?) to me re-implementing this for inclusion in the main repo?

When you have lots of inputs with long processlists it makes a big difference to break the lists up with some color for easyier viewing and also to the size on page. 

[image]
 

In the process of copying an old emoncms.org account over to a self-hosted server

Paul-Reed, 2017-11-17:
I've noticed the same, and currently it's harder to visualise processes, as they look so similar, so yes, gets my vote!
I assume you will try and establish a protocol, such as all conditional processes = green, all mathematical processes = blue, etc, etc.

Paul

pb66, 2017-11-17:
_"I assume you will try and establish a protocol,"_

Try being the operative word there, when we did it last time the code was very different and the process module or ability to drop in additional processing modules didn't exist, so I do not currently know whats possible, but I thought I would ask the question before investing loadsa time in it.

Thanks for the reply, hopefully other will feel the same way.

pb66, 2017-11-17:
Not wanting to get ahead of myself, but what has also become apparent whilst solely creating processlists for the last 5hrs solid is that a "quick link bar" showing ALL the processes as colored buttons as an alternative to using the drop down selector would make things soooo much faster.

a quick mock up with only a handful of process "quick link buttons" to convey the idea

[image]

I'm not suggesting we get rid of the dropdown box, just adding "quick links", even if only for the most frequently used processes.

TrystanLea, 2018-07-10:
The add process quick link bar is a nice idea!

TrystanLea, 2018-07-10:
For @emrysr benefit, this is how the processes used to be defined (and are still defined on emoncms.org) alongside colour coding https://github.com/emoncms/emoncmsorg/blob/master/emoncms/Lib/tablejs/custom-table-fields.js#L70

emrysr, 2018-07-16:
new issue raised for the "quick links"

https://github.com/emoncms/emoncms/issues/926

closed 2018-08-01 by TrystanLea