emoncms/emoncms #750: "ERROR: Could not save processlist. Invalid characters in arg" when

moafrancky, 2017-12-05:
Hi,

In input process list, I cannot publish to  MQTT when I use a '/' character.

Publish to 'power1' works
Publish to 'emon/power1' brings 'ERROR: Could not save processlist. Invalid characters in arg' when I want to save.

Kind regards,

My configuration
```

Emoncms | Version | 9.8.24 \| 2017.11.27
-- | -- | --
  | Modules | Administration \| App v1.0.0 \| Dashboard v1.1.1 \| EventProcesses \| Feed \| Graph v1.0.0 \| Input \| notify \| CoreProcess \| Schedule \| Time \| User \| Visualisation
Server | OS | Linux 4.9.0-4-amd64
  | Host | emoncms  (127.0.1.1)
  | Date | 2017-12-05 19:13:57 CET
  | Uptime | 19:13:57 up 21:04, 2 users, load average: 0.00, 0.00, 0.00
HTTP | Server | Apache/2.4.25 (Debian) HTTP/1.1 CGI/1.1 80
MySQL | Version | 5.5.5-10.1.26-MariaDB-0+deb9u1
  | Host | localhost (127.0.0.1)
  | Date | 2017-12-05 19:13:57 (UTC 01:00‌​)
  | Stats | Uptime: 75853 Threads: 1 Questions: 18965 Slow queries: 0 Opens: 22 Flush tables: 1 Open tables: 16 Queries per second avg: 0.250
Redis | Version | 3.2.6
  | Host | localhost:6379 (127.0.0.1)
  | Size | 48 keys (800.23K)Flush
  | Uptime | 0 days
MQTT | Version | 1.4.14
  | Host | localhost:1883 (127.0.0.1)
Memory | RAM | Used 42.23%Total: 474.12 MB Used: 200.21 MB Free: 273.91 MB
  | Swap | Used 0.00%Total: 510 MB Used: 0 B Free: 510 MB
Disk | Mount | Stats
  | / | Used 9.95%Total: 19.07 GB Used: 1.9 GB Free: 16.18 GB
PHP | Version | 7.0.19-1 (Zend Version 3.0.0)
  | Modules | apache2handler \| calendar v7.0.19-1 \| Core v7.0.19-1 \| ctype v7.0.19-1 \| curl v7.0.19-1 \| date v7.0.19-1 \| dio v0.0.9 \| dom v20031129 \| exif v1.4 \| fileinfo v1.0.5 \| filter v7.0.19-1 \| ftp v7.0.19-1 \| gettext v7.0.19-1 \| hash v1.0 \| iconv v7.0.19-1 \| igbinary v2.0.1 \| json v1.4.0 \| libxml v7.0.19-1 \| mcrypt v7.0.19-1 \| mysqli v7.0.19-1 \| mysqlnd \| openssl v7.0.19-1 \| pcre v7.0.19-1 \| PDO v7.0.19-1 \| pdo_mysql v7.0.19-1 \| Phar v2.0.2 \| posix v7.0.19-1 \| readline v7.0.19-1 \| redis v3.1.1 \| Reflection v7.0.19-1 \| session v7.0.19-1 \| shmop v7.0.19-1 \| SimpleXML v7.0.19-1 \| sockets v7.0.19-1 \| SPL v7.0.19-1 \| standard v7.0.19-1 \| sysvmsg v7.0.19-1 \| sysvsem v7.0.19-1 \| sysvshm v7.0.19-1 \| tokenizer v7.0.19-1 \| wddx v7.0.19-1 \| xml v7.0.19-1 \| xmlreader v7.0.19-1 \| xmlwriter v7.0.19-1 \| xsl v7.0.19-1 \| Zend OPcache v7.0.19-1 \| zlib v7.0.19-1 \|

```

TrystanLea, 2017-12-06:
Thanks for spotting, looks like there needs to be a / in the validation here:
https://github.com/emoncms/emoncms/blob/master/Modules/input/input_model.php#L532
I will get this fixed and tested

moafrancky, 2017-12-07:
Great, thanks for your quick feedback.

TrystanLea, 2017-12-08:
This is now fixed in the master branch https://github.com/emoncms/emoncms/commit/39461b4dc4b383c4ed74a7707237c82875b3315b

moafrancky, 2017-12-09:
It works fine. Thanks

closed 2017-12-09 by moafrancky