emoncms/emoncms #752: Input process errors

spike3312, 2017-12-06:
Hi,

Having upgraded to the latest version from 9.8.8, i now get the following error quite a bit trying to add input processors:

ERROR: Could not save processlist. undefined

I get this trying to add send an email to certain feeds but not all, also when i try to add tags such as {value} to any email. Ones already configured are fine but if i try and edit the process list i then cannot save it without getting the error above.

Matt

TrystanLea, 2017-12-06:
Hello Matt, could you check to see if there are any further details on errors printed in your browsers developer tools under the network tab? There may be more information in the response to process/set?

spike3312, 2017-12-07:
Hi @TrystanLea 

Many thanks for the reply.

Should have thought of that myself as I've been using it on one of my own pages to catch errors.

I get the following results trying to edit an already existing input processor just changing the text in the email settings:

`Object { id: "4404", description: "", name: "vcc", nodeid: "Conservatory", processList: "1:124,35:Enviro/Conservatory/V,49:3.6,52:10,eventp__ifnotmuteskip:3600,52:10,eventp__sendemail:Conservatory temperature sensor battery is {value} at {time}", time: 1512630087, value: 4.33 }
view:371:2
contextid=4404
process_ui.js:23:5
contextprocesslist=1,124,35,Enviro/Conservatory/V,49,3.6,52,10,eventp__ifnotmuteskip,3600,52,10,eventp__sendemail,Conservatory temperature sensor battery is {value} at {time}
process_ui.js:24:5
Empty string passed to getElementById().
jquery-1.11.3.min.js:2:24487
Empty string passed to getElementById().
jquery-1.11.3.min.js:2:24487
contextid=4404
process_ui.js:23:5
contextprocesslist=1,124,35,Enviro/Conservatory/V,49,3.6,52,10,eventp__ifnotmuteskip,3600,52,10,eventp__sendemail,Conservatory temperature sensor battery is {value} at {time}
process_ui.js:24:5
Empty string passed to getElementById().
jquery-1.11.3.min.js:2:24487
Empty string passed to getElementById().`

Matt.

TrystanLea, 2017-12-08:
I think the 'Empty string passed to getElementById().' is quite normal. The issue might be related to the use of / in Enviro/Conservatory? and therefore could be the same as https://github.com/emoncms/emoncms/issues/750

TrystanLea, 2017-12-08:
This may now be fixed with latest addition to master branch https://github.com/emoncms/emoncms/commit/39461b4dc4b383c4ed74a7707237c82875b3315b

spike3312, 2017-12-09:
I've not had issues with the publish to MQTT not allowing the / before as i have several setup to publish to places such as 'Enviro/Outside/H', 'Enviro/Outside/C', 'Enviro/Outside/Lux' etc. These are all working fine.

The issue still persists when trying to add send emails and using { or } characters to add {value} etc.

Matt.

spike3312, 2017-12-10:
HI,

I am also getting the following error when trying to add an input process to an input created with MQTT.

ERROR: Could not save processlist. undefined

The console shows:

contextid=0           process_ui.js:23:5
contextprocesslist=          process_ui.js:24:5
Object { id: "4739", nodeid: "Lights", name: "Downhalltable", description: "", processList: "", time: 1512873043, value: 0 }          view:371:2
contextid=4739           process_ui.js:23:5
contextprocesslist=          process_ui.js:24:5
Empty string passed to getElementById().           jquery-1.11.3.min.js:2:24487
contextid=4739            process_ui.js:23:5
contextprocesslist=1,207           process_ui.js:24:5
Empty string passed to getElementById().

Matt.

barche, 2017-12-13:
I have manually applied the patch from commit 39461b4 to the file in `/var/www/emoncms`, and this fixes the error when adding "Publish to MQTT" to an input, but the error still happens when trying to add a "Publish to MQTT" process to a virtual feed.

barche, 2017-12-14:
OK, so the same change is needed here:
https://github.com/emoncms/emoncms/blob/master/Modules/feed/feed_model.php#L989

But after that, the feed value doesn't show in MQTT

glynhudson, 2017-12-18:
Could issue be a duplication or related to: https://github.com/emoncms/emoncms/issues/541

spike3312, 2017-12-18:
Indeed it looks like the same issue, yes.

spike3312, 2017-12-20:
Could it be a permissions issue with the MQTT input script not being allowed access to the MYSQL? 

I am still getting this issue and still not getting emails when an input is updated via MQTT.  I've been throught the PHP code and as far as i can see there should be no difference between inputs updated between the HTTP API and MQTT.

glynhudson, 2017-12-21:
We have just pushed what we hope is a fix for this issue to the Emoncms master branch. See above ^^

The issue was the connection to MYSQL was timing out causing could not save process list errors. The fix will be merged into the stable branch in the next few days if no issues are reported. If you are able to please switch to Master branch and test the latest version to confirm the issue has been fixed

spike3312, 2017-12-27:
Hi, I've updated to the newer version, 9.8.27

I now get this error trying to create a new feed via MQTT:

Stack trace:
#0 /var/www/emoncms/Modules/input/input_model.php(50): exceptions_error_handler(8, 'Undefined varia...', '/var/www/emoncm...', 50, Array)
#1 /var/www/emoncms/scripts/phpmqtt_input.php(256): Input->create_input(1, 'test', 'test')
#2 [internal function]: message(Object(Mosquitto\Message))
#3 /var/www/emoncms/scripts/phpmqtt_input.php(112): Mosquitto\Client->loop()
#4 {main}
2017-12-27 23:20:20.318|ERROR|phpmqtt_input.php|exception 'ErrorException' with message 'Undefined variable: id' in /var/www/emoncms/Modules/input/input_model.php:50
Stack trace:
#0 /var/www/emoncms/Modules/input/input_model.php(50): exceptions_error_handler(8, 'Undefined varia...', '/var/www/emoncm...', 50, Array)
#1 /var/www/emoncms/scripts/phpmqtt_input.php(256): Input->create_input(1, 'test1', 'test2')
#2 [internal function]: message(Object(Mosquitto\Message))
#3 /var/www/emoncms/scripts/phpmqtt_input.php(112): Mosquitto\Client->loop()
#4 {main}

closed 2018-06-14 by glynhudson