emoncms/emoncms #763: Possible XSS in "My Account"

jna380, 2017-12-29:
Hello,
it is more a beauty mistake than really harmful, but when logged in as any user, JavaScript code can be executed through "Name", "Location", "Bio" and "Starting Page" fields in the My Account page (see appended picture).
Still the input should be sanitized. It is already prevented by "escape()" of the input in list.js.

Regards,
jna380

[image]

elyobelyob, 2017-12-30:
Nicely volunteered.

TrystanLea, 2018-05-22:
This could reuse XSS check in dashboard module:
- AntiXSS in dashboard module: https://github.com/emoncms/dashboard/tree/master/AntiXSS
- AntiXSS check in dashboard module: https://github.com/emoncms/dashboard/blob/master/dashboard_model.php#L94

emrysr, 2018-06-06:
https://github.com/emoncms/emoncms/pull/857

sanitized in list.js - php already sanitizing before db update.