emoncms/emoncms #767: Lighttpd configuration

borpin, 2018-01-04:
I have identified the lines to be added to a lighttpd.conf file if a user wishes to use that as the server for Emoncms.  I'd like to include it in the documentation.  I'd suggest a new file so it is clearly Lighttpd information and it does not then need to be repeated across different install files (linux, debian RaspberryPi).

TrystanLea, 2018-07-10:
Good idea @borpin If you would like to write this Id be happy to merge?

borpin, 2018-07-15:
Yes. I need to build a fresh instance so will do it then.

Do you want it as a separate page or part of one of the main pages?  Do you want to create a stub if a new page?

TrystanLea, 2019-07-11:
Hello @borpin I realise I did not answer your questions here, I think a separate page would be suitable, did you do this?

borpin, 2019-07-11:
Ok will do. Can you create a blank and I'll fill it in.

With Buster, and the fact Lighttpd seems to be dying through lack of interest, I'm thinking of moving to Nginx as that seems to be the way forward and is still 'lightweight' compared to Apache it seems.

TrystanLea, 2019-07-12:
Thinking more about this. ideally so as not to be trying to maintain lots of different things, that become out of date, we should be able to get rid of the raspberrypi and debian installation guides with the new build script:

- https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md
- https://github.com/emoncms/emoncms/blob/master/docs/LinuxInstall.md

A full emoncms installation is quite complex, especially where background services, redis and mosquitto are involved. I can imagine some would like to have both but Im weary of saying that we can do that given the work load involved.

How do you think a page on lighttpd would fit in?

closed 2019-11-25 by TrystanLea