emoncms/emoncms #768: DietPi Installation Documentation

borpin, 2018-01-04:
I'm working through developing a DietPi installation and I intend to document what I did/do that varies from the standard RaspberryPi install. I'd like to include it in the emoncms/docs folder as 'DietPi.md'. It is not really appropriate for the RaspberryPi folder as DietPi is not specific to the RPi.

Is that a suitable location?

Paul-Reed, 2018-01-04:
It may be better to put the guide in it's own folder, and then call it `readme.md` so the path would be `https://github.com/emoncms/emoncms/tree/master/docs/DietPi/readme.md`
Git will then recognize the readme.md file as a document, display it better, and open it whenever the folder is opened. Also, you could add any image files, or other future DietPi material in the same folder - keeping everything together.

Paul

borpin, 2018-01-04:
Could do, although the other files in the docs folder have not followed that convention.  However, as there may well be tweaks required for different types of installs (so more documents), perhaps it makes more sense.  I think I'd have a readme.md and then a file with (in the first instance) install instructions for a VM.  I'll ponder it. I don't want to create the docs in such a way that the same information is repeated as then you have to remember to update the same thing in multiple places.

closed 2018-06-14 by glynhudson