emoncms/emoncms #769: Displaying the admin page details on the forum

pb66, 2018-01-07:
(ref the [No graphs in dashboard following update](https://community.openenergymonitor.org/t/no-graphs-in-dashboard-following-update/6163?u=pb66) forum thread)

I noticed the users attempt to use a table to display the data from the admin page, which would be a nice way of doing it.

| | | |
|--- | --- | --- |
| Emoncms | Version | 9.8.27 \| 2017.12.21 |
||Modules|Administration \| App v1.0.0 \| Backup v1.1.1 \| Dashboard v1.1.1 \| EventProcesses \| Feed \| Graph v1.1.0 \| Input \| CoreProcess \| Schedule \| Time \| User \| Visualisation|

However, to achieve the above I have had to escape all the vertical separator lines used in the Admin page formatting as the markdown code uses the same character for defining the table columns, eg

```
| | | |
|--- | --- | --- |
| Emoncms | Version | 9.8.27 \| 2017.12.21 |
||Modules|Administration \| App v1.0.0 \| Backup v1.1.1 \| Dashboard v1.1.1 \| EventProcesses \| Feed \| Graph v1.1.0 \| Input \| CoreProcess \| Schedule \| Time \| User \| Visualisation|
```

If we used a different separator (eg a colon?) it would be much easier to use a markdown table on the forum (and in documentation?)

| | | |
|--- | --- | --- |
| Emoncms | Version | 9.8.27 : 2017.12.21 |
||Modules|Administration : App v1.0.0 : Backup v1.1.1 : Dashboard v1.1.1 : EventProcesses : Feed : Graph v1.1.0 : Input : CoreProcess : Schedule : Time : User : Visualisation|

Having now seen the same data using a colon rather than a vertical line in the example above, I also now feel it is much easier to read the latter, the former now appearing quite busy in comparison so I think a change of separator might improve the admin page regardless.

A "copy to clipboard" button might be useful and able to include the vertical line separators and prefix with the markdown for the table header to enable direct pasting to the forum.

thib66, 2018-01-12:
Hello, please see #775.
This PR should solve the request concerning "copy to clipboard" button

manicmoddin, 2018-01-13:
Hi,

Personally, I prefer the | separator. It looks much clearer to me.

It's a personal thing though

Jimmy

thib66, 2018-01-13:
Hello,
@pb66, @manicmoddin please see the following topic on the forum : 
https://community.openenergymonitor.org/t/copy-system-info-to-clipboard/6232
regards,
Aymeric

glynhudson, 2018-01-17:
> system-info-to-clipboard/6232

I personally prefer the separator. Just looks a bit clear.

Agree, a copy to clipboard would be awesome.

It would also be great to be able to have a single button to download all
the the log files: emoncms, emonhub, update and the admin configuration so
we could capture all info we need to debug in one do. Currently so much
time is wasted with support requests whole asking for all the info we need.

- sent from my mobile device

thib66, 2018-01-28:
Hello,
@pb66, I think that your request is satisfied thanks to the last improvements made. 
Please, could you close the issue?
Thanks,
Regards,
Aymeric

glynhudson, 2018-03-18:
Yes, good idea. Thanks guys 👍

closed 2018-03-18 by glynhudson