emoncms/emoncms #770: zoom.php not working

cmgj74, 2018-01-08:
Since about Feb 2017 zoom has been broken.

I think I found the fix:

This line:
  get_feed_data_DMY_async(vis_feed_kwh_data_callback,null,kwhd,start,end,"**day**"); // get 5 years of daily kw_data

should be:

  get_feed_data_DMY_async(vis_feed_kwh_data_callback,null,kwhd,start,end,"**daily**"); // get 5 years of daily kw_data

I saw this fixed here once...but it looks like it was lost.
[https://github.com/emoncms/emoncmsorg/commit/5900e4afbe09bfb39c6291cbf8c8c358f5eba299#diff-94c2ec2f418ea0a1e4f47bc37a905cfb](url)

cmgj74, 2018-01-09:
I have tested this in my lab, and it fixes the graph the way it used to work.

pb66, 2018-01-09:
Hi, thanks for debugging and reporting. I suspect this was only ever fixed on the emoncmsorg repo. I've tried the same fix here and it works as expected. Thanks @cmgj74 

ref https://github.com/emoncms/emoncms/commit/ab4190283d466ed0d53b383211d2e25467cdedd7#diff-e443846d35461a3175ed193ec6afdbaa

closed 2018-01-09 by pb66

pb66, 2018-01-09:
Turns out the simplezoom was also broken with the same issue fixed in ab4190283d466ed0d53b383211d2e25467cdedd7

cmgj74, 2018-01-09:
Thanks for fixing!