emoncms/emoncms #777: Non Numeric date strings not processed correctly if older date/time

borpin, 2018-01-13:
It appears to me that Emoncms currently looks like it processes the 'time' passed in non numeric string format when it does not if that date/time is not 'now'.

If a non numeric time value is passed as a string, that is close to time now, the data gets processed implying the date/time field has been accepted.  If the time is set to less than now (e.g. 10s), the data is not processed and the input is shown as n/a. No idea why a string close to 'now' gets through; those are my observations.

If a time in seconds is used, that is less than now, it is processed correctly.

This behavior has been tested on the Input API and was observed as part of the modifications to the phpmqtt_input work I am doing.

I have put a test into the phpmqtt_input to check and convert a string date, but this should be more deeply embedded.  I attempted to do this, but it seems the integer casting of time objects in other places may be causing problems later on (a guess I admit).

[From PHP docuemtation](http://us2.php.net/strtotime)
````php
if (is_string($time)){
    if (($timestamp = strtotime($time)) === false) {
        $log->warn("Time string not valid ".$time);
    } else {
        $time = $timestamp;
    }
}
````

TrystanLea, 2018-02-14:
Hello Brian, Im not sure that I follow, where does a non numeric time values processed with strtotime come in to emoncms previous to your pull request?

borpin, 2018-02-14:
Hi Trystan, I picked up from this thread https://community.openenergymonitor.org/t/uri-not-acceptable/5928/6 that there was a perception that the API accepted an ISO format time string.  This appears to be confirmed by the API not rejecting a string value if it is 'now' but anything older does get rejected (input simply does not appear).

If the API is not designed to accept a string (and looking at the code this is doubtful), it should at least not accept a string or produce a warning.

Looking at this again, I think bulk could be left as it is, but changing the main API is I think quite simple
I think they key line is this one, and doing something similar to the above code should work.
https://github.com/emoncms/emoncms/blob/c400a14e319a154c8bb8901640ad7ff9f4032ca7/Modules/input/input_methods.php#L56

However, I don't know enough about the overall code structure to be sure.

HTH, Brian

TrystanLea, 2018-02-15:
Im not entirely sure why a date time string that is equivalent to the present moment worked ok, I would have thought all strings will convert to 0 by the integer type casting.

> If the API is not designed to accept a string (and looking at the code this is doubtful), it should at least not accept a string or produce a warning.

I think I would prefer this rather than accepting a new format.

TrystanLea, 2018-02-16:
I've merged in your json mqtt_input pull request which includes the ISO string time format. Id be happy to disable the ISO string format support for consistency for the time being?

closed 2018-07-10 by TrystanLea