emoncms/emoncms #787: Unit field in the feeds table

pb66, 2018-02-05:
Could we consider adding a user defined unit field to the feeds table? A simple smallish text field would suffice, this would allow the correct units to be displayed throughout emoncms, on the graphs page and in dashboard widgets. 

This has come up before, but I raised this issue as I was reviewing the changes to the datapoint hover over in [emoncms/graphs PR#20](https://github.com/emoncms/graph/pull/20), it would be nice to see the units in the hover-over, without hardcoding to "w" or "kWh" as has been done elsewhere.

It would also add so much value to the data being saved, currently we rely on assumption (eg energy must be kWh and power must be watts), memory or ugly feed name that contain a hint. If the unit was held in a field of the feed table it could also be queried via api or even included in some of the fuller api responses.

(PS - This data could easily be pre-filled via the device module)

TrystanLea, 2018-06-06:
The first part of this has now been added thanks to @emrysr https://github.com/emoncms/emoncms/pull/855

TrystanLea, 2018-06-06:
It would now be great to automatically set the unit based on the input processes used e.g power_to_kwh would result in a feed with the unit set to kwh

emrysr, 2018-06-06:
added °C to units list:
https://github.com/emoncms/emoncms/pull/859

pb66, 2018-06-06:
Hi Emrys, Will it be possible to use free text units as well as (or instead of) selecting from a predefined list?

Just to name a few more litres, cubic meters, gallons, £, $ (and many other currencies, both minor and major units) £/ph (plus other currencies, units and periods) ppm, mph, kph, mm/day (plus other increments and periods) mV, mA, VA, Mb, Gb (and other units of disk/ram usage) pulses events dB Hz seconds millis etc etc just to name a few off the top of my head.

pb66, 2018-06-07:
Thanks Emrys, I just took a look at the feed units, great addition!

[image]

are you aware there is a blank space at the top of the dropdown? and some border or background shading might improve the list appearance a tad.

[image]

emrysr, 2018-06-08:
Thanks @pb66 

The blank space is intentional - it handles the case where the user hasn't filled in (or doesn't want) the unit.

what browser are you seeing the dropdown border in? I think the width of the select elements is wider than the select itself in the the above screenshot..

this is what it looks like in Firefox
[image]
 

I'll take a look.

pb66, 2018-06-10:
I'm using latest/updated chrome, I assume the black, white and orange colour scheme a Firefox thing? I have seen another of [your posts showing another dropdown](https://github.com/emoncms/emoncms/pull/858) and that is also shown as black and orange where as that has always been white, black and blue on my installs. The black, white and orange would stick out like a sore thumb on any of my installs (viewed via chrome).

Here is that same export interval dropdown as currently displayed on emoncms.org in the chrome browser. You can see there is a border line on the RHS of the dropdown list and the one on the LHS is not as heavy as the unit dropdown's LHS, it's almost as if the new unit dropdown has zero space between the LHS and RHS border causing a double line on the LHS before the content ( I doubt that is the case, I'm just using that as a way of describing what I see)

[image]

glynhudson, 2018-06-14:
Looking ok for me on Chrome Version 66.0.3359.181 (Official Build) (64-bit) Ubuntu 16.04

[image]

glynhudson, 2018-06-14:
Can anyone else re-create this issue? @pb66 could you try running Chrome in Incognito mode to eliminate any browser extensions? What version Chrome are you using and what OS and screen resolution?

pb66, 2018-06-17:
It looks like it might have been a browser cache issue as a crtl-f5 seems to have (partially) cleared it.

I say "partially" as the dropdown does now have a righthand border but sometimes the bottom border is missing. See the following 4 screenshots, all taken from the same page without refreshing or changing anything, just clicking the dropdown, the 1st one the bottom of the dropdown is missing although the list is complete and the page is plenty big enough

[image]

Then it's there for the 2nd and 3rd feeds

[image]   [image]

then it's missing again in the 4th

[image]

This appears to be pretty random, altering the vertical scroll, refreshing or resizing the page seems to get different results each time, but they are never all complete.

This is a small point and I'm not particularly fussed about it, it was just an observation. I certainly don't want to to take focus away from more important stuff.

I'm using Chrome Version 67.0.3396.87 (Official Build) (64-bit) on Windows 10 (fully updated). I've tried it with a laptop screen as well as a 3840x2160 monitor at various zoom levels.

Also note that it seems to be possible to open several feeds for editing at once, I'm not sure that is desirable.

emrysr, 2018-06-18:
thanks for all the detail. I'll refer to your screenshots this when I next look at this.

closed 2018-07-11 by TrystanLea