emoncms/emoncms #791: "Error creating user" at the first register on a fresh installtion

belba, 2018-02-08:
If I follow the installation instruction for Ubuntu (https://github.com/emoncms/emoncms/blob/master/docs/LinuxInstall.md) I meet the point where I have to make the first reistration for the admin-user.

At this point I got only the "Error creating user" message. No further hints in any logfile I could found (apache, mysql,syslog).

I played also with $stmt->errno and $stmt->error statements around line 225 in Modules/user/user_model.php - with no any success.

The problem is also in discuss at the community forum from Opennergymonitor:
https://community.openenergymonitor.org/t/emoncms-error-creating-user/6490

any ideas?

belba, 2018-02-12:
@geoffrey2312 
please check if in the table "users" the column "tags" has default null. If not try following SQL command:

ALTER TABLE `users` CHANGE `tags` `tags` TEXT CHARACTER SET utf8 COLLATE utf8_general_ci NULL DEFAULT NULL;

More abot this issu you can find under 
[https://community.openenergymonitor.org/t/emoncms-error-creating-user/6490](url)

geoffrey2312, 2018-02-12:
Huzzah - Worked perfectly thank you.

MatthiasMattiza, 2018-02-17:
Having the same problem. worked perfect! Thanks a lot.

pb66, 2018-02-17:
The problem should have been fixed by PR #796 and this issue closed then, can you please confirm you have had the issue with code cloned/downloaded/updated in the last 4 days?

If yes please provide more details on your setup as we will need to look at this again.

If no I think we can close this issue.

geoffrey2312, 2018-02-17:
I had this issue on Monday I spun up an Amazon EC2 with Ubuntu server and had thus issue.

TrystanLea, 2018-06-04:
Im not 100% sure that this is still a fixed issue..

If I change the user_schema from:

    'tags' => array('type' => 'text', 'default'=>NULL),

to:

    'tags' => array('type' => 'text', 'default'=>NULL, 'Null'=>true),

and recreate the tags column it seems to work again on my laptop test system. It does work fine however on an emonpi, without this change

TrystanLea, 2018-06-04:
I've created a pull request with my proposed changes https://github.com/emoncms/emoncms/pull/852

glynhudson, 2018-06-14:
@geoffrey2312 @pb66 could you check that PR #852 fixes this issue for you? Should it be merged?

closed 2018-06-26 by TrystanLea

takkaria, 2018-07-19:
I'm still experiencing this issue.  Looks like a bug in dbschemasetup.php.

**History of the bug**

It looks like in commit c42b615c, the user schema 'tags' field got `'Null'=>true` where it didn't exist before.  (Maybe as part of debugging?)

This was then removed in 30a336c8 and replaced with `'default'=>Null`.  This was the same as not including a default field at all, because the default for an unfilled field is NULL already.

Then in 2c913bff, it got `'Null'=>true` back.

**The bug**

The bug is that when creating a table, the logic that decides whether a field is going to get a NOT NULL constraint is the following line in `dbschemasetup.php`:

```php
if ($null=="NO") $query .= " not null";
```

This runs afoul of PHP's bizarre equality operator, which means that when `$null` is `true`, `$null=="NO"` evaluates to true:

```
jjquark:~ takkaria$ php -a
php > echo (true == "NO");
1
```

The simple solution to this bug is to remove `'null'=>true` from user_schema.php.  However, when investigating this bug, I started reworking the DB schema setup code because at the moment there are three sets of logic for deciding how fields should be configured (when adding a new table, when adding a new field to an existing field, and when altering an existing field) which means that it's quite inconsistent.  I'll open a pull request for this patch.

PR #852 (2018-06-26): add null=true and print mysql error
Fixes #791