emoncms/emoncms #795: Restructure data and settings paths

Paul-Reed, 2018-02-10:
### restructure feature branch
The aim of this feature branch is to change the structure of emoncms, so that it accords to the Filesystem Hierarchy Standard, and enables most of the data/settings files to be housed in a single parent folder or partition, to facilitate easier backup routines, RO/RW filesystems and keep the settings file out of the html path.

The changes made so far are;
- The three data folders have been grouped under a single parent folder - /var/lib/emondata/
- Changes made to feed engine paths in default.settings file.
- Move the settings file to a new folder - /etc/emoncms/
- Addition made to process_settings.php to ensure backwards compatibility.
- Build guide updated to reflect above changes.

These changes only affect users building a new system from the guide and are transparent to existing systems being updated via 'git pull' (although optionally, existing systems can be easily updated).

If the changes so far are supported, the next stage which would mainly apply to SD card builds, would be to replace the /var/lib/emondata & /etc/emoncms folders with symlinks, and move the actual folders to /home/data. Further discussion then about adding mysql & emonhub config to the same folder. This would then result in all emoncms user files being stored in just one folder - /home/data.
The /home/data folder could equally be an external drive, USB stick or RW partition.

For consideration

Paul

TrystanLea, 2018-07-10:
For future reference, when we make this change it would be a good time to rename the phpmqtt_input script as both effect the emonpi install and update process https://github.com/emoncms/emoncms/issues/915

biva, 2018-08-06:
Hello,
I’m new to emoncms and I just installed it on Debian 9.5
I’d be interested in moving the data folder to another location. What’s the status of this discussion? Did you chose the best method?
Thanks!

Paul-Reed, 2018-08-21:
Forum discussion link; https://community.openenergymonitor.org/t/move-data-settings-file-to-a-common-location/6464

TrystanLea, 2018-08-22:
I've updated your restructure branch @Paul-Reed to include latest commits from the master branch, a cleaner comparison is now possible again: https://github.com/emoncms/emoncms/compare/restructure

What was the rationale for moving the /var/lib/emondata folder rather than /var/lib/phpfina etc?  Is that for symlinking/mounting on a separate partition/drive?

TrystanLea, 2018-08-22:
The changes look overall quite harmless and backwards compatible as you say. Users using /var/lib/phpfina would need to be aware of the emondata folder addition if creating a new settings file from default.settings.php

Paul-Reed, 2018-08-22:
> What was the rationale for moving the /var/lib/emondata folder rather than /var/lib/phpfina etc? Is that for symlinking/mounting on a separate partition/drive?

Yes. It would enable the phpfina, phpfiwa etc folders to be contained within a single parent folder - emondata, which keeps all of the emoncms data files together. This would result in the data folders being clearly identified as emoncms data files, make symlinking clearer and also simplify data backups/restores.

TrystanLea, 2019-07-11:
I think this issue can now be closed.

We are moving to a hierarchical settings approach as described in this issue: https://github.com/emoncms/emoncms/issues/1346

and the new emonSD build script places the emoncms data directories in /var/opt/emoncms https://github.com/openenergymonitor/EmonScripts/blob/master/install/config.ini#L54

closed 2019-07-11 by TrystanLea