emoncms/emoncms #824: Logger Scripts and Instructions have the wrong install paths

brucemiranda, 2018-04-21:
If you follow the instructions to install emoncms on Ubuntu it tells you to install the git in /var/www/html/emoncms

But the logger instructions and the scripts are still referencing /var/www/emoncms

Same issue inside the install.sh too. And also the chown command is setting the ownership to pi

Paul-Reed, 2018-04-21:
Are you able to submit a PR, so we could look at the changes which you are suggesting please.

Paul

brucemiranda, 2018-04-21:
I am still trying to unravel the world of pain I've gotten into because I stupidly ran the install.sh without checking it first. It's got some pretty nasty sudo commands in there that just assumes you had absolutely nothing else running on the server before emoncms was installed. e.g. sudo rm /etc/rc.local

glynhudson, 2018-06-14:
Thanks for brining this up, I wonder how best logger [install.sh](https://github.com/emoncms/emoncms/blob/master/scripts/logger/install.sh) could be made to handle different Emoncms install locations? The emonPi still uses `/var/www/emoncms` as the install location. Maybe the script could be modified to ask the user to define the install path of Emoncms, or it could be autodetected or defined in `settings.php`

TrystanLea, 2019-01-17:
Pull request merged last year, thanks @brucemiranda  https://github.com/emoncms/emoncms/commit/3a9cf14c0a82c090200601cfaf50dae52fc00d03 closing this issue.

closed 2019-01-17 by TrystanLea