emoncms/emoncms #836: Breaking api changes in php-redis

pb66, 2018-05-03:
Whilst helping a user on the OEM forum I noticed that PECL was installing v4.0.2 and read the [changelog](https://pecl.php.net/package-changelog.php?package=redis&release=4.0.2).

The changelog states "*** WARNING! THIS RELEASE CONTAINS BRAKING API CHANGES! ***" for all versions above 4 and all I could find on those changes is this issue https://github.com/phpredis/phpredis/issues/1333 which has a couple of interesting linked issues.

@TrystanLea or @chaveiro are you able to confirm if these changes are going to impact emoncms?

Can we continue to use the latest phpredis versions, now at 4.x.x without changes?

Or can the required changes be made to emoncms?

Or do we need to stick with 3.x.x (currently 3.1.6) and only get critical updates moving forward?

There is always a concern about how long devs will maintain a backwards compatible version and it also presents a "hiccup" in the installation guide as we will need to define a version (eg pecl install redis-3.1.6) instead of leaving it open (eg pecl install redis) and then that needs maintaining in both the guide and any future install or update scripts.

I hope for option 1, or 2 if possible, I think 3 will start to reintroduce the issues we have not long seen resolved over v2.2.5 vs >v2.2.7 and again with php5 vs php7 prior to v3.1.3.

The apt-get package is still at [3.1.1 for Stretch](https://packages.debian.org/stretch/php-redis) which is prior to the rounding issue fix we saw in 3.1.3 and 2.2.5 for Jessie which is prior to the memory leak fix in v2.2.7 (current emonSD image version) so I think we do still need to stick with PECL.

I am unable to test v4.0.0 right now as both my test and live instances are currently on the same server so I cannot use on "test" without the risk of breaking "live".

Any thoughts?

chaveiro, 2018-05-04:
From the [changelog](https://pecl.php.net/package-changelog.php?package=redis&release=4.0.2) i think the only major issue with emoncms might be when a connection fails to establish to redis and the new exception that may not be correctly catched by current code.
Please try to update to 4.x and then will deal with any issues that arrive from that.

pb66, 2018-10-12:
See https://community.openenergymonitor.org/t/redis-version-4-safe/8813?u=pb66

TrystanLea, 2019-01-17:
We are currently running version 4.1.1 stable on the latest SD card without any known issues. I will close this for now.

closed 2019-01-17 by TrystanLea