emoncms/emoncms #854: Allow forward slash in feed process argument

mattwire, 2018-06-05:
Per the change implemented for input process argument here https://github.com/emoncms/emoncms/pull/753/files#diff-a28cf4ba30f0c22c88290de9ccddbe49R532

We need to do the same in feed process otherwise it's not possible to set the topic for "Publish to MQTT" feed process.

pb66, 2018-06-05:
Hi Matt, I assume you're referring to the virtual feed processes?

Specifically the
```
if (preg_replace('/[^\p{N}\p{L}_\s.-]/u','',$arg)!=$arg) 
```
line at https://github.com/emoncms/emoncms/blob/9.8.30/Modules/feed/feed_model.php#L989.

The feed processes are only run/updated/processed when the feed data is queried, are you wanting a value to be published each time a visualization is viewed/refreshed rather than when the data is input?

I think certain processes (including the publish to mqtt) should NOT be readily available to the virtual feed processing, so whilst you are right that the regex prevents proper use of the publish to mqtt process, I question whether that process should be accessible at all from the feeds processing as it wouldn't function as expected.

mattwire, 2018-06-05:
@pb66 That makes sense, I haven't actually got it to work despite making this change - so in fact we should probably be removing the "Publish to MQTT" feed option?

TrystanLea, 2018-06-06:
Good point Publish to MQTT should not be available as it calls Publish to MQTT repeatedly for every data point in the view! same should go for sending emails

pb66, 2018-06-06:
As of commit https://github.com/emoncms/emoncms/commit/8225e5374b21ad954897ed5c7e4e77e9a439d80b the forward slash has been added so I will change the title from "Allow forward slash in feed process argument" to "mqtt and email processes should be unavailable to feed processing" to suit the additional point.

TrystanLea, 2018-06-07:
@chaveiro do you have any recommendations as to how to restrict the processes that are available for virtual feeds?

cagabi, 2018-06-08:
Hi,

I faced the same issue when implementing processes for the Tasks module.

The processes are added to the select in the _init_ method of the _processlist_ui_ object in here https://github.com/emoncms/emoncms/blob/master/Modules/process/Views/process_ui.js#L679

In order to not display the method _Send email_ and _Publish to MQTT_ you could replace

`if (processlist_ui.contexttype == 1 && processlist_ui.processlist[z]['feedwrite'] == true) {`

with something like

`if (processlist_ui.contexttype == 1 && processlist_ui.processlist[z]['feedwrite'] == true && processlist_ui.processlist[z][0] != "Publish to MQTT" && processlist_ui.processlist[z][0] != "Send Email") {`

I am sure you can find a more elegant solution ;)

closed 2018-06-14 by glynhudson

emrysr, 2018-06-15:
issue #882 created to re-show the sendEmail and Publish to MQTT on input processes.

pull request submitted #883