emoncms/emoncms #861: Enforce unique tag:name for feeds

TrystanLea, 2018-06-07:
At present unique tag:names are enforced at feed creation but not subsequently when tags and names are edited in the feed list. The function set_feed_fields does not check for the unique requirement https://github.com/emoncms/emoncms/blob/master/Modules/feed/feed_model.php#L741

pb66, 2018-06-07:
_"At present unique tag:names are enforced at feed creation"_ it seems that is not the case (on emoncms.org at least), I just tried it because in my test for the  ([Problem setting up EmonCMS Sync - PHP errors](https://community.openenergymonitor.org/t/problem-setting-up-emoncms-sync-php-errors/7526/28?u=pb66)) forum discussion, I had indeed just edited the feed name in the feeds page, but the 3rd "solar" feed shown below was created from the inputs page with the name "solar" when there were already 2 feeds of the same name with the same tag.

[image]

TrystanLea, 2018-06-07:
exists_feed_tag function https://github.com/emoncms/emoncms/blob/master/Modules/feed/feed_model.php#L96

emrysr, 2018-06-12:
https://github.com/emoncms/emoncms/pull/867

closed 2018-06-14 by glynhudson