emoncms/emoncms #865: header 'Content-Type : text' has no sub type

mbruinenberg, 2018-06-08:
The lack of mime sub types in the header response results in issues handling EmonCMS responses at client side.

In Input.php all header content types of type text are now set to 'text'.
This must be changed to the correct mime type 'text/plain'.

emrysr, 2018-06-12:
Hi @mbruinenberg, thanks for posting this issue.

Are you reading or posting to the the inputs when you receive this "text" content type?
Can you let me know how you call the inputs to get this "text" mime in the headers? 
Are you using one of the methods listed in the (api help section) [http://localhost/emoncms/input/api]
eg: http://localhost/emoncms/input/get

When I try the above I get "application/json" as the "Content-Type" header value...

Let me know how I can re-produce your problem so that I can take a look.

borpin, 2018-06-12:
@emrysr - This community post may help.
https://community.openenergymonitor.org/t/emoncms-ubuntu-16-apache-2-4-php-7-input-post-issue-with-java-client/7600

emrysr, 2018-06-12:
great - thanks!

emrysr, 2018-06-12:
added the mime sub type text/plain to the input/post api endpoint: 
https://github.com/emoncms/emoncms/pull/873

This should fix any client issues with unrecognized content types. Will need to run more tests before I can change the content type to "application/json" - just in case it breaks other features.

thanks @borpin and @mbruinenberg

closed 2018-07-10 by TrystanLea