emoncms/emoncms #876: fixed height top menu dropdown

emrysr, 2018-06-14:
The setup dropdown menu has a fixed height. If you have too many items it it not obvious on a mobile that you can scroll down to see the rest

Removing the max-height property below would make the dropdown as large as needed.

```
/*     /Theme/basic/emon-blue.css         */
.scrollable-menu {
    height: auto;
    max-height: 250px;
    overflow-y: auto;
}
```

This height restriction must be there to prevent the dropdown from getting too long.

emrysr, 2018-06-14:
if this height restriction is required then you could increase it by 1/2 a line so that you see that the list continues below the cutoff.

glynhudson, 2018-06-14:
Thanks @emrysr . This issue is most apparent on Android mobile devices, the scroll bar is not shown therefore it's impossible for a user to know that there are additional menu items hidden below

[image]

Same menu on desktop 

[image]

emrysr, 2018-06-14:
made the dropdown taller (333px)

https://github.com/emoncms/emoncms/pull/877

PR #877 (2018-06-14): increased the dropdown max-height fixes #876
it should now allow for 3 more items before cropping off. Fixes #876
commit f3ddb0ae (2018-06-14): Merge pull request #877 from emrysr/feature/issue-876-dropdown-menu-max-height

increased the dropdown max-height fixes #876
closed 2018-06-14 by glynhudson