emoncms/emoncms #878: Add "first created" date to feeds

glynhudson, 2018-06-14:
It would be useful to be able to see when a feed was created and arrange current feeds by created date.

I don't think we currently store feed created date? Do you think this would be a good thing to add @TrystanLea?

JonMurphy, 2018-06-14:
I'd like to add the phpfina interval to this also.  This is what I use:

`sudo find /home/pi/data/phpfina/*.meta -exec sh -c 'echo "{}" ; od -i "{}" ; echo' \;`

in the results below, the 4th number is the interval.  I think the 5th number is the start epoch.

```
pi@emonpi(rw):phpfina$ sudo find /home/pi/data/phpfina/*.meta -exec sh -c 'echo "{}" ; od -i "{}" ; echo' \;
/home/pi/data/phpfina/109.meta
0000000           0           0         120  1466826960
0000020

/home/pi/data/phpfina/10.meta
0000000           0           0          10  1436400810
0000020

/home/pi/data/phpfina/110.meta
0000000           0           0         120  1466827080
0000020

/home/pi/data/phpfina/111.meta
0000000           0           0         120  1467151320
0000020

/home/pi/data/phpfina/112.meta
0000000           0           0         120  1467151320
0000020

/home/pi/data/phpfina/12.meta
0000000           0           0          10  1436401840
0000020
```

TrystanLea, 2018-06-26:
Yes, I agree somehow having access to this without over crowding the interface would be great

PR #910 (2018-07-26): Feature/issue 655 clear all old feed data
fixes: #655 
also fixes: #878

this allows for clearing all feed data and "trimming" data up to a given date

[image]
column marked yellow = clear
column marked green = trim

**NOT YET TESTED ON A EMONPI.**
closed 2018-07-26 by TrystanLea