emoncms/emoncms #888: Incorrect "en-EN" locale code used in emoncms?

pb66, 2018-06-18:
I am trying to understand some user profile code in emoncms and noticed that the default locale code seems to be "en_EN" is that intended/correct?

It doesn't seem to be a standard locale code, should we not be using "en-GB" as the default? 

(I'm new to JS and not great with PHP so apologies if I ask a stupid question)

Whilst looking at using a user account timezone in client run javascript I found the JS toLocaleString() function uses a locale to format the date and time string eg

```
var acct_time = new Date().toLocaleString("en-GB", {timeZone: "Australia/Sydney", timeZoneName: "long"})
document.getElementById('acct_time').innerHTML = acct_time;
```

so I was just pondering on whether the session[`lang`] would be suitable/available to use (in addition to adding a new session['timezone'] variable) and found this code which threw me a bit.

https://github.com/emoncms/emoncms/blob/9.8.30/locale.php#L62-L93

This won't effect using a user account (session?) timezone directly, but I was thinking that displaying the current time in both session and browser timezones might be useful to the viewer.

emrysr, 2018-06-25:
Hi @pb66 ,

I added the session variables here:
https://github.com/emoncms/emoncms/pull/896/files#diff-3a8983b5e260461fe91544f3ff97de77R733

Not yet sure what will make use of them?

pb66, 2018-06-26:
When I was trying to figure out how to make the user account timezone available to dashboards and apps etc, I noticed the "lang" variable was already in the session data and I figured that was the way to go. I have since worked out that because dashboards and apps are also available without being logged in, either publicly or using read or write apikeys in the urls, this will not work for all viewing instances.

So I now do not know if that data is of any real value in the session data as any fix for non-logged in viewing will still be available for logged in viewing, I do not know why the "lang" was already there either.

But thanks for fixing the en-GB locale code, I haven't tried it yet but it looks good :-)

closed 2018-06-26 by pb66